Koenkk/zigbee2mqtt #8748: Entities become disabled after device rename

Masterz69, 2021-09-14:
## What happened
And again about _linkquality & _voltage.

As discussed in #8490 - "These entities are disabled by default so that is expected".
I've enabled them on selected devices and expected they WILL STILL enabled forever.

But, after device got renamed (!) entities again appears disabled.

## What did you expect to happen
IMHO they can be created as disabled, but no more status toggle later. Only by user initiative.

## How to reproduce it (minimal and precise)
Just renamed devices.

## Debug info
Zigbee2MQTT version: 1.21.1-1 
Adapter information: not relevant, IMHO.

Koenkk, 2021-09-14:
Home Assistant does not support renaming, Zigbee2MQTT first undiscovers the entities and discovers them again so HA will see them as new entities and disables some entities by default. Not something that can be fixed from z2m.

Masterz69, 2021-09-15:
> Home Assistant does not support renaming, Zigbee2MQTT first undiscovers the entities and discovers them again so HA will see them as new entities and disables some entities by default. Not something that can be fixed from z2m.

As far I see - entities disablement in HA is done by PR #7684 in that particular project.
IMHO it means entities going to be disabled from Z2M side/code by f.e.:
* voltage: {device_class: 'voltage', enabled_by_default: false, state_class: 'measurement'}

So, it broken here - why not to fix right here.
F.e. invent setting - "no disabled by default" entities.
Or allowing users to select/unselect "enabled/disabled by default" on property/entity basis.

Masterz69, 2021-09-15:
In other words:
1. not "HA ... disables some entities by default", but Z2M force HA to do it;
2.  renamed device from my perspective not a NEW device.

frenck, 2021-09-20:
While that might not be your perspective, from the HA side of the spectrum, it is a new device.

Masterz69, 2021-09-20:
> While that might not be your perspective, from the HA side of the spectrum, it is a new device.

Sure. But major is HA user perspective, who spent time to enable entities, and then they again become disabled and required to restart HA instance - not good IMHO.

Z2M option 'not send disabled-by-default' or HA MQTT integration option 'ignore disabled-by-default' - possible solutions.
Or just (IMHO better way) - keep all properties/entities/... enabled (as it was) and let HA system user to decide.

frenck, 2021-09-20:
The changed behavior is the expected and wishes for behavior in Home Assistant.
I don't see the error.

Masterz69, 2021-09-20:
Someone wished to re-enable entities and restart HA core on device rename. 
As well as restart HA core on every new device - just because want to use linkquality & voltage. Who ? :)

BTW Voltage is not only battery voltage on battery-powered devices, but also AC voltage on AC-powered devices.
Why not to disable current, power, something else ? F.e. because lot of people have no idea - what current is...

frenck, 2021-09-20:
I'm having problems understanding your last comment, and thus fall to see the relation is has with the "issue" raised.

Masterz69, 2021-09-20:
Sorry my bad English, not my native language.

When I'm writing some code - not using var1, var2, var3, but some user-friendly names. As most of developers, I hope.
In same way - I'm not using IEEE addresses in automations, lovelaces etc. as device/entity names.

After that change, every device rename (because of moving to other place or replacing other device) make me troubles - I need to re-enable entities and restart HA core. What potentially make more troubles - interrupt automations, postpone timers, etc.

I'm planning to use _voltage from power outlets. And now I need to restart HA core on every new outlet added to system, because _voltage is disabled. It's a pain.

Again sorry for my English.