Koenkk/zigbee2mqtt #8749: Add _TZ3210_9q49basr as a manufacturer for TS0501B

fzsombor, 2021-09-14:
**Join warning:** Received message from unsupported device with Zigbee model 'TS0501B' and manufacturer name '_TZ3210_9q49basr'

**It came from here:** https://www.aliexpress.com/item/1005002495077553.html?spm=a2g0s.9042311.0.0.e9e74c4daUAtvq

danieledwardgeorgehitchcock, 2021-09-14:
Please read / populate / submit the correct New Device Request template for the above device

BDskii, 2021-11-14:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0501B', manufacturerName: '_TZ3210_9q49basr'}], 
        model: 'TS0501B', 
        vendor: 'TuYa', 
        description: 'Single color LED controller', 
        extend: extend.light_onoff_brightness()
    };

module.exports = definition;
```

**Information about the device + link**
Purchased here:
(https://aliexpress.ru/item/1005001865233095.html?spm=a2g39.orderlist.0.0.130c4aa6NEBFaV)

TyYa TS0501B

**data/database.db entry of the device**
`{"id":9,"type":"Router","ieeeAddr":"0xa4c138ff11ee17c8","nwkAddr":18392,"manufId":4417,"manufName":"_TZ3210_9q49basr","powerSource":"Mains (single phase)","modelId":"TS0501B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0501B","manufacturerName":"_TZ3210_9q49basr","powerSource":1,"zclVersion":3,"appVersion":91,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":91,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1636896239826}`