Koenkk/zigbee2mqtt #8753: Request to roll-back PR #7684

Masterz69, 2021-09-15:
## What happened
After mentioned PR - enabled attributes/entities in HA become disabled on device rename in Z2M.
Renamed device not a NEW device, IMHO. 

Voltage sensors widely used for battery control for battery-powered sensors.
Additionally it not only show battery voltage, but main power Voltage for AC-powered sensors, like plugs etc.

Linkquality allow to filter Z2M devices/entities - there is only way now to filter them out in Lovelace f.e.

IMHO users will choose on they own - which attributes/properties not usable in their specific circumstances.

## What did you expect to happen
NO "disabled by default" properties or, at least, configuration option to disable that 'staff'.

danieledwardgeorgehitchcock, 2021-09-15:
Isn't this discussion already happening in this issue raised by you? #8748

Why not consider renaming that thread and keeping the conversation in one place?

Masterz69, 2021-09-15:
> Isn't this discussion already happening in this issue raised by you? #8748
> 
> Why not consider renaming that thread and keeping the conversation in one place?

It also possible.
Probably it is professional deformation. Usually report each incident separate, as well problems and issues separately, even they closely related.

Koenkk, 2021-09-15:
Dupe of #8748 indeed.

closed 2021-09-15 by Koenkk