Koenkk/zigbee2mqtt #8764: Won't start when data on other directory (ZIGBEE2MQTT_DATA) and compiled with --production

XLDom, 2021-09-16:
Hello

## What happened
Z2M won't start when data on other directory (ZIGBEE2MQTT_DATA) and compiled with npm ci --production

## What did you expect to happen

## How to reproduce it (minimal and precise)
use : `npm ci --production`
start with data on other directory and use : ZIGBEE2MQTT_DATA="/mydatadir" npm start

## Debug info
Log : 
> zigbee2mqtt@1.21.1 start /opt/zigbee2mqtt
> node index.js
Building Zigbee2MQTT... (initial build), failed
(node:21043) UnhandledPromiseRejectionWarning: Error: Command failed: npm run build
sh: 1: tsc: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! zigbee2mqtt@1.21.1 build: `tsc && node index.js writehash`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.21.1 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-09-16T08_27_43_558Z-debug.log
at ChildProcess.exithandler (child_process.js:308:12)
at ChildProcess.emit (events.js:315:20)
at maybeClose (internal/child_process.js:1048:16)
at Socket.<anonymous> (internal/child_process.js:439:11)
at Socket.emit (events.js:315:20)
at Pipe.<anonymous> (net.js:673:12)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:21043) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:21043) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Koenkk, 2021-09-18:
Can you check if this issue is still present in the latest dev?  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

XLDom, 2021-09-19:
> Can you check if this issue is still present in the latest dev? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Yes, it's same issue.
It's when i do : `npm ci --production`

here is the log : 
```
> zigbee2mqtt@1.21.1-dev start /opt/zigbee2mqtt
> node index.js
Building Zigbee2MQTT... (hash changed), failed
(node:8693) UnhandledPromiseRejectionWarning: Error: Command failed: npm run build
sh: 1: tsc: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! zigbee2mqtt@1.21.1-dev build: `tsc && node index.js writehash`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.21.1-dev build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-09-19T20_44_47_787Z-debug.log
at ChildProcess.exithandler (child_process.js:390:12)
at ChildProcess.emit (events.js:400:28)
at maybeClose (internal/child_process.js:1055:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:8693) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:8693) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
if i do a simple : `npm ci`
it's running.

Koenkk, 2021-09-20:
I've checked, but is expected that starting with `npm ci --production` doesn't work, this is because this won't install the required typescript compiler.

XLDom, 2021-09-21:
Do you have a solution ?

Koenkk, 2021-09-21:
No, this is as intended, why do you need `npm ci --production`?

XLDom, 2021-09-22:
> No, this is as intended, why do you need `npm ci --production` instead of `npm ci`?

no reason lol. I've see that into a documentation... i don't know where ^^
I will use npm ci now.
Thanks

closed 2021-09-22 by Koenkk