Koenkk/zigbee2mqtt #8769: *Another* Template Variable Warning

OzGav, 2021-09-16:


danieledwardgeorgehitchcock, 2021-09-16:
This might have been broken in this release of Home Assistant:

[2021.4](https://www.home-assistant.io/blog/2021/04/07/release-20214/#breaking-changes)

OzGav, 2021-09-18:
Maybe so. Either way I am hoping to work out how to resolve this.

mwav3, 2021-10-01:
The templates in Home assistant need a "default" condition defined now.  I set my own custom template to clear the error and posted it here for the Hampton bay fan https://github.com/Koenkk/zigbee2mqtt/issues/8617

I'm not sure how to fix the template sent during mqtt discovery by zigbee2mqtt,  otherwise I would submit a pr.

The concern is that starting in 2021.4 this is a warning, but per those release notes, will error in a coming release, possibly as soon as 2021.10

So if this isn't fixed, eventually the template will stop working on some future release.

OzGav, 2021-10-03:
Thanks. Let me try your templatre and see if that removes my warnings and then we can see if we can work out what has to happen to get a PR done.

Koenkk, 2021-10-09:
Can you check if this has been fixed in the latest-dev? (edge for hassio)

I see you are still using 1.18.1, probably you are using the old repository; check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

OzGav, 2021-10-10:
Yes no warnings in the logs and everything seems to be working correctly (I am getting a warning about units changing from V to mV for my Xiaomi buttons but that seems correct and understandable so can sort that out myself)

Thanks @Koenkk