Koenkk/zigbee2mqtt #8770: Fifthplay smart switch

Mariusthvdb, 2021-09-16:
really sorry if this can t be more helpful, but I am at a loss. Ive used Fifthplay smart switches on a Delta Comfortwijzer hub for years. However, having moved, and not being able to keep the energy company's hub, I am stuck with the extra plugs I bought for this Hub.

They are Zigbee's alright, or so I am told, but there is no way I can include them, or even see any response in the Z2M panel.

this is the one: https://www.manualslib.com/manual/815298/Fifthplay-Energysmart.html?page=27#manual or http://public.fifthplay.com/manuals/HandleidingEnergySmart_NL.pdf

I understand this is not much to go on, but would appreciate if you could at least help me out getting these devices to talk to Z2M.

only thing I can think of, is that Spider device uses the same plugs for which they were licensed. and still available: https://www.bol.com/nl/nl/p/slimme-stekker-spider-powerplug/9200000050793981/

Hope you can find any lead to follow up on this,
thanks!
Marius
## Information about the device + link
..

## data/database.db entry of the device
..

danieledwardgeorgehitchcock, 2021-09-16:
I have done a bit of googling, and there appears to be a lot of conflicting information on various sites about Fifthplay devices.

Some sites just say 'Radio Open Protocol', the site you mentions above mentions 'Zigbee' (however, that was back in 2009, so could be TouchLink), quite a few sites though, seem to suggest that these devices might be 'Z-Wave'.

If you are unable to get any debug messages through Zigbee2MQTT, and these devices have been unbound from their native router, that would suggest to me that the protocol that they use probably isn't Zigbee..

danieledwardgeorgehitchcock, 2021-09-16:
It looks like these devices communicate on Z-Wave:

[Z-Wave Alliance](https://products.z-wavealliance.org/products/2190?selectedFrequencyId=1)

Mariusthvdb, 2021-09-16:
hmm, I dont think so. In the time I used it, I talked to their service desk about Z-wave, and they provided a dongle to add to the hub required for being able to use that protocol.

But, its easy, I can check, I use Z-wave myself too ;-)

I did try touchlink in the Z2M add-on, but it reported the red window.

let me get back on the z-wave..

closed 2021-11-25 by github-actions[bot]