Koenkk/zigbee2mqtt #8775: No such file or directory (homeassistant last version) after 24h

malosaa, 2021-09-17:
##where are you using the device
Home Assistant OS 9.6 (virtualbox vm)
## What happened
After 24 hours i get this message
```
Zigbee2MQTT:error 2021-09-17 09:45:50: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
[09:45:51] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.21.1 start
> node index.js
```
error in console:
```
[19747.492579] ohci-pci 0000:00:06.0: frame counter not updating; disabled
[19747.493188] ohci-pci 0000:00:06.0: HC died; cleaning up
```

If i disconnect it trough virtualbox and connect it again i get this message:
```
71:38:47.450810 VUSB: Detached '0000000004e16e60[proxy 0451:16a8]' from port 2 on RootHub#0
71:39:17.642619 VUSB: Disabled '0000000004e172e0[proxy 0451:16a8]' remote wakeup for configuration 0
71:39:17.643185 VUSB: Attached '0000000004e172e0[proxy 0451:16a8]' to port 2 on RootHub#0 (FullSpeed)
```
As still i get the above error, i always need to full reboot vm host to work again.

If i don't reboot host and use the windows script to run Z2M it works directly.. 

## How to reproduce it (minimal and precise)
It just happens after 24h

## Debug info
Zigbee2MQTT version: 1.21.1-1
Adapter hardware: CC2531, 
Adapter firmware version: i tried all versions..

Koenkk, 2021-09-18:
I believe this issue is connected to https://github.com/Koenkk/zigbee2mqtt/issues/8282, please keep an eye on that issue and try the latest-dev once #8282  is solved.

Koenkk, 2021-09-18:
Update: I believe it is the same as #8376

closed 2021-09-18 by Koenkk

Drakot, 2023-12-14:
This is happening to me, and i checked the issues you mentioned and no luck updating
 The issue is probably with virtualbox
 Any help?