Koenkk/zigbee2mqtt #8777: Lidl HG06463 version B?

RM-Them, 2021-09-17:
Hello!

I think I have a variation of the Lidl HG06463 (B version perhaps?) which is a larger bulb. 

Here is the two side by side, unfortunately this was the only image I could find : https://static01.eu/1offers.co.uk/images/uploads/030721/livarno-lux-large-smart-filament-bulb43645.jpg

Before the bulb was supported, I added a js file which allowed both the slimmer bulb and the larger bulb to work:

`const definition = {
    zigbeeModel: ['TS0501A'], 
    model: 'HG06463', 
    vendor: 'Lidl', 
    description: 'Lidl Smart LED Filament Bulb', 
    extend: extend.light_onoff_brightness(),
};`

With the support added, something went a bit wrong and I had to remove the .js file as it caused issues with the module to load. Now, the slimmer bulb works fine, but the larger one is unrecognised.

I'm a bit new to all this, but I guess a new entry with a new manufacturerName is needed? In my dashboard it reports as:

Zigbee Model : TS0501A
Zigbee Manufacturer : _TZ3000_nosnx7im 

Let me know if I missed anything or I need to give any more info.

Thanks!

danieledwardgeorgehitchcock, 2021-09-17:
Hi,

I added the slimmer bulb to the codebase so, I should be able to help you out.

The bulbs should have slightly different model numbers - do you still have the box? It will be in the smallprint on the back..

Could you also post the database.db entry for the larger bulb?

RM-Them, 2021-09-17:
Hey!

The Model number on the back of the box is HG06463B.

The JSON item for the device from the .db seems to be

```json
{
    "id": 3,
    "type": "Router",
    "ieeeAddr": "0xbc33acfffe0d9587",
    "nwkAddr": 5183,
    "manufId": 4098,
    "manufName": "_TZ3000_nosnx7im",
    "powerSource": "Mains (single phase)",
    "modelId": "TS0501A",
    "epList": [
        1,
        242
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 257,
            "inClusterList": [
                0,
                3,
                4,
                5,
                6,
                8,
                768,
                4096
            ],
            "outClusterList": [
                25,
                10
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TS0501A",
                        "manufacturerName": "_TZ3000_nosnx7im",
                        "powerSource": 1,
                        "zclVersion": 3,
                        "appVersion": 81,
                        "stackVersion": 0,
                        "hwVersion": 1,
                        "dateCode": ""
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "242": {
            "profId": 41440,
            "epId": 242,
            "devId": 97,
            "inClusterList": [],
            "outClusterList": [
                33
            ],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 81,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1629309392579
}
```

Just for future knowledge, would it be the case that I would just need to submit a new PR to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lidl.js

```javascript
{
    fingerprint: [{modelID: 'TS0501A', manufacturerName: '_TZ3000_nosnx7im'}],
    model: 'HG06463B',
    vendor: 'Lidl',
    description: 'Livarno Lux E27 ST64 filament bulb',
    extend: extend.light_onoff_brightness({disableEffect: true}),
    meta: {turnsOffAtBrightness1: false},
}
```

or would other steps need to be taken?

Thanks again for your help and let me know if I can help further!

danieledwardgeorgehitchcock, 2021-09-20:
> Just for future knowledge, would it be the case that I would just need to submit a new PR to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lidl.js

Yes, that is all that would be required here :-)

tlvince, 2021-11-25:
Support for this was added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3332. I also found a [product description page](https://www.lidl.de/p/livarno-lux-leuchtmittel-spiralfilament-dimmbar-zigbee-smart-home/p100311870) (in German).