Koenkk/zigbee2mqtt #8780: Issue adding a new device

tanker11, 2021-09-17:
Hello,

I am trying to add a new device according to the descriptions, but when logging in debugging mode, the following is what I get:

"Device with modelID 'undefined' is not supported."

The device has no buttons for on/off, but a power mains, so I cannot "nudge" it to tell more about itself.

Is there any suggestion how I should continue?
The model is a Sunricher SRP-ZG9105-50CCT device.

Thanks in advance

danieledwardgeorgehitchcock, 2021-09-17:
This sound like the device interview hasn't completed properly.

How far away is your device from the coordinator when you are trying to get it to join your network?

tanker11, 2021-09-17:
I have about two meters from the coordinator, which is a CC2652P with
antenna. Is it too far?

The connected lamp flashes during the pairing and goes off, showing it is
successful, but indeed the interview may be incomplete.

Can I check shomewhere in the logs how far it goes?

2021. szept. 17., Pén 16:46):

tanker11, 2021-09-18:
Hello,

I moved closer to 10cm from the coordinator, and recorded the debug log, see below.
Manufacturer and modelID are still 'Undefined'.
How could I find out if this is a problem of the device  or problem of communication between the coordinator or anything else?

warn  2021-09-18 09:21:56: Device 'furdozigbee' left the network
info  2021-09-18 09:21:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"furdozigbee","ieee_address":"0x000d6f000fc7512b"},"type":"device_leave"}'
info  2021-09-18 09:21:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x000d6f000fc7512b"},"type":"device_removed"}'
info  2021-09-18 09:22:01: Device 'furdozigbee' joined
info  2021-09-18 09:22:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"furdozigbee","ieee_address":"0x000d6f000fc7512b"},"type":"device_joined"}'
info  2021-09-18 09:22:01: Starting interview of 'furdozigbee'
info  2021-09-18 09:22:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"furdozigbee","ieee_address":"0x000d6f000fc7512b","status":"started"},"type":"device_interview"}'
info  2021-09-18 09:22:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"furdozigbee"},"type":"device_connected"}'
info  2021-09-18 09:22:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"furdozigbee"},"type":"pairing"}'
debug 2021-09-18 09:22:11: Device 'furdozigbee' announced itself
info  2021-09-18 09:22:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"furdozigbee","ieee_address":"0x000d6f000fc7512b"},"type":"device_announce"}'
info  2021-09-18 09:22:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"furdozigbee"},"type":"device_announced"}'
debug 2021-09-18 09:23:26: Saving state to file /app/data/state.json
info  2021-09-18 09:26:35: Successfully interviewed 'furdozigbee', device has successfully been paired
warn  2021-09-18 09:26:35: Device 'furdozigbee' with Zigbee model 'undefined' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2021-09-18 09:26:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"furdozigbee","ieee_address":"0x000d6f000fc7512b","status":"successful","supported":false},"type":"device_interview"}'
info  2021-09-18 09:26:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"furdozigbee","supported":false},"type":"pairing"}'