Koenkk/zigbee2mqtt #8782: Zigbee2Mqtt is not publishing homeassistant discovery topics after update to 1.21.1

krzysztofdrobek, 2021-09-17:
## What happened

Zigbee2mqtt devices were not available to home assistant. Upon further inspection I found out that there is no messages on homeassistant topic. After downgrading to version 1.21.0 everything works correctly. 

## What did you expect to happen

Zigbee2mqtt should publish home assistant discovery topics.

## How to reproduce it 

1. Upgrade to 1.21.1
2. Check if there are messages sent to homeassistant topic

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC2531

coco673, 2021-09-18:
I've the same issue and for me the pairing of device is not working too

krzysztofdrobek, 2021-09-18:
> I've the same issue and for me the pairing of device is not working too

Reverting to 1.21.0 and updating back to 1.21.1 fixed the issue for me.

closed 2021-09-18 by krzysztofdrobek