Koenkk/zigbee2mqtt #8788: Delay in reporting device status since upgrading to 1.21.1

Plawasan, 2021-09-17:
Since moving to 1.21.1 I'm seeing a notable delay between toggling a physical switch and its status update message appearing in z2m console (and ultimately in HA). This has been happening with multiple switches (all Tuya TS0001 or TS0002), I tried to re-pair them but it had no noticeable effect. Related but not quite on topic - I also tried to use bind to link them together and completely bypass z2m however I simply cannot figure out how to do that (or if it's even possible), 
```
service: mqtt.publish
data:
  topic: zigbee2mqtt/bridge/request/device/bind
  payload: '{"from": "wardrobe_switch", "to": "bedroom_switch/l1","clusters": ["genOnOff"]}'
```
results in 
```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"from": "wardrobe_switch", "to": "bedroom_switch/l1","clusters": ["genOnOff"]}'
Error Nothing to bind from 'wardrobe_switch' to 'bedroom_switch'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"from":"wardrobe_switch","to":"bedroom_switch"},"type":"device_bind_failed"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"wardrobe_switch","to":"bedroom_switch/l1"},"error":"Nothing to bind","status":"error"}'
Error Nothing to bind
```

Back to the original issue... toggling the switch state from HA or from Z2M frontend has absolutely no delay and works 100%, the delay is only between using the actual switch and the reporting of status change.

## What happened
Device status is reported with a delay of 1-10s

## What did you expect to happen
Device status reported immediately (<1s)

## How to reproduce it (minimal and precise)
Trigger the physical button, watch for relevant update to appear in z2m consolse.

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: ZZH! CC2652R
Adapter firmware version: Coordinator firmware version: '{"meta":"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'

Koenkk, 2021-09-18:
Could be due to https://github.com/Koenkk/zigbee2mqtt/issues/8663

Plawasan, 2021-09-18:
I am on 5.10.60-v7+ so I guess I'll just wait for .62.. thanks.

Still baffled by the bind situation :)

closed 2021-09-18 by Koenkk

Plawasan, 2021-09-30:
For the record - I have used rpi-update to get to  5.10.63-v7+ 

https://github.com/Hexxeh/rpi-firmware/tree/64132d67d3e083076661628203a02d27bf13203c

pi@raspberrypi:~ $ uname -r
5.10.60-v7+
pi@raspberrypi:~ sudo rpi-update 64132d67d3e083076661628203a02d27bf13203c
..magic and reboot..
pi@raspberrypi:~ $ uname -r
5.10.63-v7+

The response is now pretty much immediate - probably better than ever before. Problem hopefully solved.