Koenkk/zigbee2mqtt #8789: Error: Unrecognized nodeId:27117

dumbodumbos, 2021-09-17:
The following error occurs periodically

(node:349) UnhandledPromiseRejectionWarning: Error: Unrecognized nodeId:27117
    at Driver.networkIdToEUI64 (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:454:19)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at EZSPAdapter.processMessage (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:50:33)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:349) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
(node:349) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

z2m latest version. Xiaomi 3 gateway in z2m mode with integration https://github.com/AlexxIT/XiaomiGateway3

dumbodumbos, 2021-10-01:
Integration developer comment

This is a known issue with z2m. Please ask the z2m project to modify the EZSP driver to ensure stable operation in the mode of a large number of errors from the Zigbee chip. Unfortunately, this gateway has a lot of them.

For example, the ZHA driver is much more robust.

Koenkk, 2021-10-02:
Please add your issue here: https://github.com/Koenkk/zigbee-herdsman/issues/319. Note that the ezsp driver is in a very experimental and therefore issues are expected. I recommend to use on of the adapters mentioned on https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html instead.

closed 2021-10-02 by Koenkk

slayer, 2021-12-02:
Same problem with Gateway 3
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[00:09:36] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[00:09:36] INFO: MQTT available, fetching server detail ...
[00:09:37] INFO: Previous config file found, checking backup
[00:09:37] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[00:09:37] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:09:37] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Assertion failed
Assertion failed
Assertion failed
Zigbee2MQTT:warn  2021-12-03 00:09:47: Using experimental new availability feature
Error: Unrecognized nodeId:45754
    at Driver.networkIdToEUI64 (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:454:19)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at EZSPAdapter.processMessage (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:50:33)
[00:09:49] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Assertion failed
Assertion failed
Assertion failed
Zigbee2MQTT:warn  2021-12-03 00:09:58: Using experimental new availability feature
Error: Unrecognized nodeId:45754
    at Driver.networkIdToEUI64 (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:454:19)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at EZSPAdapter.processMessage (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:50:33)
[00:10:03] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Assertion failed
Assertion failed
Assertion failed
Zigbee2MQTT:warn  2021-12-03 00:10:12: Using experimental new availability feature
```