Koenkk/zigbee2mqtt #8790: Changing brightness state results in 10 second timeout, but the command is accepted.

domcx, 2021-09-17:
## What happened
After pairing about 3 Sengled bulbs, they appear to be reported correctly and can be turned on and off. However, changing the brightness of the bulb results in command timeout. After this the bulb is not usable for about 10s while it waits for he command response.

```
zigbee2mqtt     | Zigbee2MQTT:debug 2021-09-17 12:56:55: Received MQTT message on 'zigbee2mqtt/0xb0ce1814030a3f5f/set' with data '{"state": "ON", "brightness": 190}'
zigbee2mqtt     | Zigbee2MQTT:debug 2021-09-17 12:56:55: Publishing 'set' 'brightness' to '0xb0ce1814030a3f5f'
zigbee2mqtt     | 2021-09-17T17:56:55.154Z zigbee-herdsman:controller:endpoint Command 0xb0ce1814030a3f5f/1 genLevelCtrl.moveToLevelWithOnOff({"level":190,"transtime":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt     | 2021-09-17T17:56:55.155Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xb0ce1814030a3f5f:62069/1 (0,0,1)
zigbee2mqtt     | 2021-09-17T17:56:55.155Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62069,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":27,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,12,4,190,0,0]}}
zigbee2mqtt     | 2021-09-17T17:56:55.155Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,117,242,1,1,8,0,27,0,30,6,1,12,4,190,0,0,14]
zigbee2mqtt     | 2021-09-17T17:56:55.168Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt     | 2021-09-17T17:56:55.168Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt     | 2021-09-17T17:56:55.168Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt     | 2021-09-17T17:56:55.168Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt     | 2021-09-17T17:56:55.168Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt     | 2021-09-17T17:56:55.173Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,27,221]
zigbee2mqtt     | 2021-09-17T17:56:55.173Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,27,221]
zigbee2mqtt     | 2021-09-17T17:56:55.173Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,27] - 221
zigbee2mqtt     | 2021-09-17T17:56:55.173Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":27}
zigbee2mqtt     | 2021-09-17T17:56:55.173Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt     | 2021-09-17T17:56:56.207Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,117,242,2,197,220,230,187,71]
zigbee2mqtt     | 2021-09-17T17:56:56.207Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,117,242,2,197,220,230,187,71]
zigbee2mqtt     | 2021-09-17T17:56:56.207Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [117,242,2,197,220,230,187] - 71
zigbee2mqtt     | 2021-09-17T17:56:56.208Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":62069,"relaycount":2,"relaylist":[56517,48102]}
zigbee2mqtt     | 2021-09-17T17:56:56.208Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt     | 2021-09-17T17:56:56.261Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,8,0,117,242,1,1,0,54,0,34,21,180,0,0,7,24,121,10,0,0,32,190,230,187,27,80]
zigbee2mqtt     | 2021-09-17T17:56:56.261Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,8,0,117,242,1,1,0,54,0,34,21,180,0,0,7,24,121,10,0,0,32,190,230,187,27,80]
zigbee2mqtt     | 2021-09-17T17:56:56.261Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,8,0,117,242,1,1,0,54,0,34,21,180,0,0,7,24,121,10,0,0,32,190,230,187,27] - 80
zigbee2mqtt     | 2021-09-17T17:56:56.261Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":62069,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":54,"securityuse":0,"timestamp":11801890,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,121,10,0,0,32,190]}}
zigbee2mqtt     | 2021-09-17T17:56:56.262Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":121,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":32,"attrData":190}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":62069,"endpoint":1,"linkquality":54,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
zigbee2mqtt     | Zigbee2MQTT:debug 2021-09-17 12:56:56: Received Zigbee message from '0xb0ce1814030a3f5f', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":190}' from endpoint 1 with groupID 0
zigbee2mqtt     | 2021-09-17T17:56:56.264Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt     | Zigbee2MQTT:info  2021-09-17 12:56:56: MQTT publish: topic 'zigbee2mqtt/0xb0ce1814030a3f5f', payload '{"brightness":190,"linkquality":54,"state":"OFF","update":{"state":"available"},"update_available":true}'
zigbee2mqtt     | 2021-09-17T17:57:05.174Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0xb0ce1814030a3f5f:62069,0)
zigbee2mqtt     | 2021-09-17T17:57:05.174Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 62069
zigbee2mqtt     | 2021-09-17T17:57:05.175Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":62069,"options":0,"radius":30}
zigbee2mqtt     | 2021-09-17T17:57:05.175Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,117,242,0,30,253]
zigbee2mqtt     | 2021-09-17T17:57:05.187Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
zigbee2mqtt     | 2021-09-17T17:57:05.187Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
zigbee2mqtt     | 2021-09-17T17:57:05.187Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
zigbee2mqtt     | 2021-09-17T17:57:05.187Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
zigbee2mqtt     | 2021-09-17T17:57:05.187Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt     | 2021-09-17T17:57:08.191Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xb0ce1814030a3f5f:62069/1 (1,0,1)
zigbee2mqtt     | 2021-09-17T17:57:08.191Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62069,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":28,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,12,4,190,0,0]}}
zigbee2mqtt     | 2021-09-17T17:57:08.191Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,117,242,1,1,8,0,28,0,30,6,1,12,4,190,0,0,9]
zigbee2mqtt     | 2021-09-17T17:57:08.204Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt     | 2021-09-17T17:57:08.205Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt     | 2021-09-17T17:57:08.205Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt     | 2021-09-17T17:57:08.205Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt     | 2021-09-17T17:57:08.205Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt     | 2021-09-17T17:57:08.209Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,28,218]
zigbee2mqtt     | 2021-09-17T17:57:08.209Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,28,218]
zigbee2mqtt     | 2021-09-17T17:57:08.209Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,28] - 218
zigbee2mqtt     | 2021-09-17T17:57:08.209Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":28}
zigbee2mqtt     | 2021-09-17T17:57:08.209Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt     | 2021-09-17T17:57:18.214Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0xb0ce1814030a3f5f:62069,1)
[... cut ...]
zigbee2mqtt     | 2021-09-17T17:57:26.765Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

## What did you expect to happen

The brightness level to change (which it does) and operation to continue like normal.

## How to reproduce it (minimal and precise)

1. Pair a Sengled bulb.
2. Send command to turn the bulb "ON" and set brightness to some level.

## Debug info
Zigbee2MQTT version: Zigbee2MQTT version 1.21.1
Adapter hardware:  CC2652P2
Adapter firmware version: Z-stack 20210708

domcx, 2021-09-17:
I tried ZHA and have no such issues.

I tried to issue OTA updates to my Sengled bulbs through Z2M and it fails saying that the device didn't respond. However i see info logs saying that the device is asking for OTA updates.

Koenkk, 2021-09-18:
Can you try pairing the bulb close to the coordinator (so that it has a direct link), does this issue still appear then?

domcx, 2021-09-18:
My config looks like this if that's relevant:
```
homeassistant: true
permit_join: true
mqtt:
  [excluded]
serial:
  port: /dev/ttyUSB0
advanced:
  pan_id: [excluded]
  channel: 24
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
device_options:
  legacy: false
```

I started from scratch and re-paired the closest bulbs. The follow is what resulted when I changed the brightness. Switching them on and off works as expected.

I should also mention that I have turned off the 2.4Ghrz bands on my APs to avoid interference as best I can.

```
zigbee2mqtt    | Zigbee2MQTT:debug 2021-09-18 12:09:17: Received MQTT message on 'zigbee2mqtt/0xb0ce1814030a3f5f/set' with data '{"state": "ON", "brightness": 183}'
zigbee2mqtt    | Zigbee2MQTT:debug 2021-09-18 12:09:17: Publishing 'set' 'brightness' to '0xb0ce1814030a3f5f'
zigbee2mqtt    | 2021-09-18T17:09:17.248Z zigbee-herdsman:controller:endpoint Command 0xb0ce1814030a3f5f/1 genLevelCtrl.moveToLevelWithOnOff({"level":183,"transtime":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt    | 2021-09-18T17:09:17.248Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xb0ce1814030a3f5f:18785/1 (0,0,1)
zigbee2mqtt    | 2021-09-18T17:09:17.248Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18785,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":44,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,42,4,183,0,0]}}
zigbee2mqtt    | 2021-09-18T17:09:17.248Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,97,73,1,1,8,0,44,0,30,6,1,42,4,183,0,0,185]
zigbee2mqtt    | 2021-09-18T17:09:17.260Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2021-09-18T17:09:17.260Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2021-09-18T17:09:17.260Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2021-09-18T17:09:17.261Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2021-09-18T17:09:17.261Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-09-18T17:09:17.264Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,44,234]
zigbee2mqtt    | 2021-09-18T17:09:17.265Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,44,234]
zigbee2mqtt    | 2021-09-18T17:09:17.265Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,44] - 234
zigbee2mqtt    | 2021-09-18T17:09:17.265Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":44}
zigbee2mqtt    | 2021-09-18T17:09:17.265Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-09-18T17:09:27.272Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0xb0ce1814030a3f5f:18785,0)
zigbee2mqtt    | 2021-09-18T17:09:27.272Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 18785
zigbee2mqtt    | 2021-09-18T17:09:27.273Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":18785,"options":0,"radius":30}
zigbee2mqtt    | 2021-09-18T17:09:27.273Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,97,73,0,30,82]
zigbee2mqtt    | 2021-09-18T17:09:27.278Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
zigbee2mqtt    | 2021-09-18T17:09:27.278Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
zigbee2mqtt    | 2021-09-18T17:09:27.278Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
zigbee2mqtt    | 2021-09-18T17:09:27.278Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
zigbee2mqtt    | 2021-09-18T17:09:27.278Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-09-18T17:09:30.279Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xb0ce1814030a3f5f:18785/1 (1,0,1)
zigbee2mqtt    | 2021-09-18T17:09:30.279Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18785,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":45,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,42,4,183,0,0]}}
zigbee2mqtt    | 2021-09-18T17:09:30.279Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,97,73,1,1,8,0,45,0,30,6,1,42,4,183,0,0,184]
zigbee2mqtt    | 2021-09-18T17:09:30.292Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt    | 2021-09-18T17:09:30.292Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt    | 2021-09-18T17:09:30.292Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt    | 2021-09-18T17:09:30.292Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt    | 2021-09-18T17:09:30.292Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2021-09-18T17:09:30.297Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,45,235]
zigbee2mqtt    | 2021-09-18T17:09:30.297Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,45,235]
[... cut ...] alse}) failed (Timeout - 18785 - 1 - 42 - 8 - 11 after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
```

I also started using a ConBee II which exhibited the same issue when used with Z2M (and many others that were unrelated). I returned that device and switched to the CC2652P2 for better support.

Koenkk, 2021-09-20:
- Do you maybe have a CC2531 to sniff the traffic when sending the command? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)
- What model sengled bulbs are these?