Koenkk/zigbee2mqtt #8791: IKEA Tradfri LED1923R5 can't not transition?

codingcatgirl, 2021-09-17:
## What happened
Somehow all of my Ikea Tradfri LED1923R5 lights always act out any state change with an about ~0.7s long transition. I am able to extend the transition time to 2, 5, etc seconds, but i am unable to set it to zero. This does not happen with my Ikea Tradfri ICPSHC24-30EU-IL-1, where transitions only happen when i explicitely request them. 

## What did you expect to happen
If the transition attribute is not set or set to zero, any changes should happen instantaneously, just as it does with the other type of Tradfri device.

## How to reproduce it (minimal and precise)
Sending any commant to the `/set` topic for one of my Tradfri spots, i get a transition that i can't seem to be able to get rid of.

## More Information
If my memory doesn't fail me, this is a new issue that the lights previously didn't have. I do remember setting a global setting in Home Assistant to enable transitions for every light state change. However, i do assume that this shouldn't affect zigbee2mqtt if i use the mqtt interface directly, should it?

I have disabled this setting in home assistant to be sure, and the result is what i described above. One type of Tradfri device can be turned off/on or dimmed without any transition, but the spot lights can't be. 

I have searched all of the settings in the Zigbee2MQTT UI (including the raw configuration dump available there) and can't find any place where a default transition is set for these devices. 

I can't rule out that this might be a hardware restriction of this specific Tradfri model, although i find that hard to believe because i feel like i remember not having transitions on this lamp a few months ago. However, if it is a hardware restriction after all and this model always enforces a transition, that would mean that the documentation for it is wrong and should probably be updated.

Please help me. Thank you <3

## Debug info
Zigbee2MQTT version: 1.21.1 (Home Assistant Add-on)

T-IOOO, 2021-10-04:
I have eight of these GU10 bulbs an they are also acting weird compared to other Trådfri bulbs. Before I was able to set the startup temp but then it disappeared. For some reason only two bulbs have the startup temp parameters in "Exposes".

One thing I also noticed is that when Ikea beta formware is activated I get error message when trying OTA update. "No image available for ImageType '10242'"

MattWestb, 2021-10-05:
IKEA test OTA feed is for the moment hosting one year old firmware's (after being down for some days) and is **not** recommended to being used for production systems.

In the production feed is all the latest OTA files but one large release is expected very soon in the test feed with nearly all controller device and blinds that have passing Zigbee certification so better not using it for not getting unexpected  problem. 

I think you is having different version of your GU10 lights but you can see that in the system.

MattWestb, 2021-10-05:
Your light is to new for the old test feed files:
<html>
<body>

zingo_lds_bulb_hwpwmcs_cws | 0x2802 10242 | WRGB Bulb | 3 ZB3 | 1.0.021 ✅ | 🔲
-- | -- | -- | -- | -- | --

</body>
</html>
Its the new CWS3 and they have getting one update after the release and your can have one older firmware from the factory.

closed 2021-11-12 by github-actions[bot]