Koenkk/zigbee2mqtt #8798: Update dont work

salopette, 2021-09-18:
andrej@Hassio:/opt/zigbee2mqtt$ ./update.sh
ERROR: Backup directory exists. May be previous restoring was failed?
1. Save 'data-backup' and 'data' dirs to safe location to make possibility to re                                               store config later.
2. Manually delete 'data-backup' dir and try again.
andrej@Hassio:/opt/zigbee2mqtt$ ./update.sh
Stopping Zigbee2MQTT...
Creating backup of configuration...
Updating...
Bereits aktuell.
Installing dependencies...
npm ERR! code EBADPLATFORM
npm ERR! notsup Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin"}                                                (current: {"os":"linux","arch":"x64"})
npm ERR! notsup Valid OS:    darwin
npm ERR! notsup Valid Arch:  undefined
npm ERR! notsup Actual OS:   linux
npm ERR! notsup Actual Arch: x64

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/andrej/.npm/_logs/2021-09-18T16_51_36_908Z-debug.log
andrej@Hassio:/opt/zigbee2mqtt$ ^C
andrej@Hassio:/opt/zigbee2mqtt$ sudo systemctl start zigbee2mqtt
andrej@Hassio:/opt/zigbee2mqtt$ sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at Sat 2021-09-18 17:36:27 CEST. --
Sep 18 18:52:59 Hassio npm[18724]:     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12) {
Sep 18 18:52:59 Hassio npm[18724]:   code: 'MODULE_NOT_FOUND',
Sep 18 18:52:59 Hassio npm[18724]:   requireStack: [ '/opt/zigbee2mqtt/index.js' ]
Sep 18 18:52:59 Hassio npm[18724]: }
Sep 18 18:52:59 Hassio systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 18:52:59 Hassio systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Sep 18 18:53:00 Hassio systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Sep 18 18:53:00 Hassio systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 2.
Sep 18 18:53:00 Hassio systemd[1]: Stopped zigbee2mqtt.
Sep 18 18:53:00 Hassio systemd[1]: Started zigbee2mqtt.
Sep 18 18:53:01 Hassio npm[18745]: > zigbee2mqtt@1.21.1-dev start
Sep 18 18:53:01 Hassio npm[18745]: > node index.js
Sep 18 18:53:01 Hassio npm[18745]: internal/modules/cjs/loader.js:892
Sep 18 18:53:01 Hassio npm[18745]:   throw err;
Sep 18 18:53:01 Hassio npm[18745]:   ^
Sep 18 18:53:01 Hassio npm[18745]: Error: Cannot find module 'semver'
Sep 18 18:53:01 Hassio npm[18745]: Require stack:
Sep 18 18:53:01 Hassio npm[18745]: - /opt/zigbee2mqtt/index.js
Sep 18 18:53:01 Hassio npm[18745]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:889:15)
Sep 18 18:53:01 Hassio npm[18745]:     at Function.Module._load (internal/modules/cjs/loader.js:745:27)
Sep 18 18:53:01 Hassio npm[18745]:     at Module.require (internal/modules/cjs/loader.js:961:19)
Sep 18 18:53:01 Hassio npm[18745]:     at require (internal/modules/cjs/helpers.js:92:18)
Sep 18 18:53:01 Hassio npm[18745]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:16)
Sep 18 18:53:01 Hassio npm[18745]:     at Module._compile (internal/modules/cjs/loader.js:1072:14)
Sep 18 18:53:01 Hassio npm[18745]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1101:10)
Sep 18 18:53:01 Hassio npm[18745]:     at Module.load (internal/modules/cjs/loader.js:937:32)
Sep 18 18:53:01 Hassio npm[18745]:     at Function.Module._load (internal/modules/cjs/loader.js:778:12)
Sep 18 18:53:01 Hassio npm[18745]:     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12) {
Sep 18 18:53:01 Hassio npm[18745]:   code: 'MODULE_NOT_FOUND',
Sep 18 18:53:01 Hassio npm[18745]:   requireStack: [ '/opt/zigbee2mqtt/index.js' ]
Sep 18 18:53:01 Hassio npm[18745]: }
Sep 18 18:53:01 Hassio systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 18:53:01 Hassio systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Sep 18 18:53:01 Hassio systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Sep 18 18:53:01 Hassio systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 3.
Sep 18 18:53:01 Hassio systemd[1]: Stopped zigbee2mqtt.
Sep 18 18:53:01 Hassio systemd[1]: Started zigbee2mqtt.
Sep 18 18:53:02 Hassio npm[18774]: > zigbee2mqtt@1.21.1-dev start
Sep 18 18:53:02 Hassio npm[18774]: > node index.js
Sep 18 18:53:02 Hassio npm[18774]: internal/modules/cjs/loader.js:892
Sep 18 18:53:02 Hassio npm[18774]:   throw err;
Sep 18 18:53:02 Hassio npm[18774]:   ^
Sep 18 18:53:02 Hassio npm[18774]: Error: Cannot find module 'semver'
Sep 18 18:53:02 Hassio npm[18774]: Require stack:
Sep 18 18:53:02 Hassio npm[18774]: - /opt/zigbee2mqtt/index.js
Sep 18 18:53:02 Hassio npm[18774]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:889:15)
Sep 18 18:53:02 Hassio npm[18774]:     at Function.Module._load (internal/modules/cjs/loader.js:745:27)
Sep 18 18:53:02 Hassio npm[18774]:     at Module.require (internal/modules/cjs/loader.js:961:19)
Sep 18 18:53:02 Hassio npm[18774]:     at require (internal/modules/cjs/helpers.js:92:18)
Sep 18 18:53:02 Hassio npm[18774]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:16)
Sep 18 18:53:02 Hassio npm[18774]:     at Module._compile (internal/modules/cjs/loader.js:1072:14)
Sep 18 18:53:02 Hassio npm[18774]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1101:10)
Sep 18 18:53:02 Hassio npm[18774]:     at Module.load (internal/modules/cjs/loader.js:937:32)
Sep 18 18:53:02 Hassio npm[18774]:     at Function.Module._load (internal/modules/cjs/loader.js:778:12)
Sep 18 18:53:02 Hassio npm[18774]:     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12) {
Sep 18 18:53:02 Hassio npm[18774]:   code: 'MODULE_NOT_FOUND',
Sep 18 18:53:02 Hassio npm[18774]:   requireStack: [ '/opt/zigbee2mqtt/index.js' ]
Sep 18 18:53:02 Hassio npm[18774]: }
Sep 18 18:53:02 Hassio systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 18:53:02 Hassio systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Sep 18 18:53:02 Hassio systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Sep 18 18:53:02 Hassio systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 4.
Sep 18 18:53:02 Hassio systemd[1]: Stopped zigbee2mqtt.
Sep 18 18:53:02 Hassio systemd[1]: Started zigbee2mqtt.
Sep 18 18:53:02 Hassio npm[18793]: > zigbee2mqtt@1.21.1-dev start
Sep 18 18:53:02 Hassio npm[18793]: > node index.js
Sep 18 18:53:02 Hassio npm[18793]: internal/modules/cjs/loader.js:892
Sep 18 18:53:02 Hassio npm[18793]:   throw err;
Sep 18 18:53:02 Hassio npm[18793]:   ^
Sep 18 18:53:02 Hassio npm[18793]: Error: Cannot find module 'semver'
Sep 18 18:53:02 Hassio npm[18793]: Require stack:
Sep 18 18:53:02 Hassio npm[18793]: - /opt/zigbee2mqtt/index.js
Sep 18 18:53:02 Hassio npm[18793]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:889:15)
Sep 18 18:53:02 Hassio npm[18793]:     at Function.Module._load (internal/modules/cjs/loader.js:745:27)
Sep 18 18:53:02 Hassio npm[18793]:     at Module.require (internal/modules/cjs/loader.js:961:19)
Sep 18 18:53:02 Hassio npm[18793]:     at require (internal/modules/cjs/helpers.js:92:18)
Sep 18 18:53:02 Hassio npm[18793]:     at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:16)
Sep 18 18:53:02 Hassio npm[18793]:     at Module._compile (internal/modules/cjs/loader.js:1072:14)
Sep 18 18:53:02 Hassio npm[18793]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1101:10)
Sep 18 18:53:02 Hassio npm[18793]:     at Module.load (internal/modules/cjs/loader.js:937:32)
Sep 18 18:53:02 Hassio npm[18793]:     at Function.Module._load (internal/modules/cjs/loader.js:778:12)
Sep 18 18:53:02 Hassio npm[18793]:     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12) {
Sep 18 18:53:02 Hassio npm[18793]:   code: 'MODULE_NOT_FOUND',
Sep 18 18:53:02 Hassio npm[18793]:   requireStack: [ '/opt/zigbee2mqtt/index.js' ]
Sep 18 18:53:02 Hassio npm[18793]: }
Sep 18 18:53:02 Hassio systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 18:53:02 Hassio systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Sep 18 18:53:03 Hassio systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Sep 18 18:53:03 Hassio systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Sep 18 18:53:03 Hassio systemd[1]: Stopped zigbee2mqtt.
Sep 18 18:53:03 Hassio systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Sep 18 18:53:03 Hassio systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Sep 18 18:53:03 Hassio systemd[1]: Failed to start zigbee2mqtt.

Bartekn86, 2021-09-18:
25 files changed, 36658 insertions(+), 2331 deletions(-)
 rename lib/extension/{bridge.js => bridge.ts} (57%)
Installing dependencies...
npm ERR! code EBADPLATFORM
npm ERR! notsup Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin"} (current: {"os":"linux","arch":"arm"})
npm ERR! notsup Valid OS:    darwin
npm ERR! notsup Valid Arch:  undefined
npm ERR! notsup Actual OS:   linux
npm ERR! notsup Actual Arch: arm

npm ERR! A complete log of this run can be found in:

salopette, 2021-09-19:
i have now switched to master branch and it works. But Dev Zweig does not work.

Bartekn86, 2021-09-19:
update works, please update nodejs to 14.x in your system

salopette, 2021-09-19:
Kak?

Bartekn86, 2021-09-19:
nodejs -v ?
curl -sL https://deb.nodesource.com/setup_14.x | sudo -E bash -

salopette, 2021-09-19:
nodejs -v
v14.17.6

Bartekn86, 2021-09-19:
strange

Bartekn86, 2021-09-19:
node -v ?

salopette, 2021-09-19:
node -v
v14.17.6

tteck, 2021-09-19:
Same proble
[2021-09-19T13_01_00_210Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/7192132/2021-09-19T13_01_00_210Z-debug.log)
m