Koenkk/zigbee2mqtt #8800: Issue with adding new device "How to support new devices."

EHVBspotter, 2021-09-18:
## What happened
I want to add a new IKEA Blub and I followed the instruction when you have a new device. I'm not able to start zigbee2mqtt and logs are rolling and looping without the needed information.

It's the following device, it's the follow of the LED1623G12
```
    zigbeeModel: ['TRADFRIbulbE27WSglobeopal1055lm'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LED2003G10', // Vendor model number, look on the device for a model number
    vendor: 'IKEA', // Vendor of the device (only used for documentation and startup logging)
    description: 'TRADFRI LED bulb E27 1055 lumen, dimmable, white spectrum, opal white', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
```

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: latest

danieledwardgeorgehitchcock, 2021-09-20:
Hi,

Please can you post the debug logs and the content of your external converter file?

This should help us narrow down the issue!

EHVBspotter, 2021-09-20:
Hi,
That's the main problem. when I follow the steps the program is looping with the log files.
What I added to my configuration.yml is the advance line and external_converters line pointing to the JS file. then after saving the yml I restarted the services and on that moment no logging at all.

see attached the files I used
[Used_files.zip](https://github.com/Koenkk/zigbee2mqtt/files/7194695/Used_files.zip)

danieledwardgeorgehitchcock, 2021-09-20:
Just had a look... You have got no extend key, or fromZigbee / toZigbee.

You need to use either of the above for it to work...

Try this:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const bulbOnEvent = async (type, data, device) => {
    /**
     * IKEA bulbs lose their configured reportings when losing power.
     * A deviceAnnounce indicates they are powered on again.
     * Reconfigure the configured reoprting here.
     * NOTE: binds are not lost so rebinding is not needed!
     */
    if (type === 'deviceAnnounce') {
        for (const endpoint of device.endpoints) {
            for (const c of endpoint.configuredReportings) {
                await endpoint.configureReporting(c.cluster.name, [{
                    attribute: c.attribute.name, minimumReportInterval: c.minimumReportInterval,
                    maximumReportInterval: c.maximumReportInterval, reportableChange: c.reportableChange,
                }]);
            }
        }
    }
};

const tradfriExtend = {
    light_onoff_brightness: (options = {}) => ({
        ...extend.light_onoff_brightness(options),
        exposes: extend.light_onoff_brightness(options).exposes.concat(e.power_on_behavior()),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),
    light_onoff_brightness_colortemp: (options = {colorTempRange: [250, 454]}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        exposes: extend.light_onoff_brightness_colortemp(options).exposes.concat(e.power_on_behavior()),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),
    light_onoff_brightness_colortemp_color: (options = {disableColorTempStartup: true, colorTempRange: [250, 454]}) => ({
        ...extend.light_onoff_brightness_colortemp_color(options),
        exposes: extend.light_onoff_brightness_colortemp_color(options).exposes.concat(e.power_on_behavior()),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),

const definition = {
    zigbeeModel: ['TRADFRIbulbE27WSglobeopal1055lm'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LED2003G10', // Vendor model number, look on the device for a model number
    vendor: 'IKEA', // Vendor of the device (only used for documentation and startup logging)
    description: 'TRADFRI LED bulb E27 1055 lumen, dimmable, white spectrum, opal white', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: tradfriExtend.light_onoff_brightness_colortemp(),
};

module.exports = definition;
```

EHVBspotter, 2021-09-20:
Yes, Thanks that did trick!!!

Thanks in advance you made my day ;)

closed 2021-09-20 by EHVBspotter

pawelbudynek, 2021-09-22:
@EHVBspotter did you manage to add LED2003G10 model into home asisstant?

EHVBspotter, 2021-09-22:
I did this in Domoticz it's self. so not in home asisstant.