Koenkk/zigbee2mqtt #8801: AssertionError [ERR_ASSERTION]: No image available for imageType '5634' at getImageMeta

mkania11, 2021-09-18:
## What happened
Ota update failed:
Update of 'Termostat lazienka' failed (No image available for imageType '5634')
Debug AssertionError [ERR_ASSERTION]: No image available for imageType '5634' at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:24:5) at runMicrotasks (<anonymous>) at processTicksAndRejections (internal/process/task_queues.js:95:5) at getNewImage (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:412:18) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:281:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.js:222:21) at Controller.callExtensionMethod (/app/lib/controller.js:421:58)
## What did you expect to happen
Firmware update
## How to reproduce it (minimal and precise)
Home assistant - Zigbee2MQTT - OTA - Update device firmware

## Debug info
Zigbee2MQTT version: Zigbee2mqtt Edge Current version: edge
Adapter hardware: TS0601_thermostat
Adapter firmware version:
unknown (not show in Zigbee2MQTT)

lorenc88, 2021-09-21:
Got the same issue with TS0601_thermostat.

vladi1234, 2021-09-23:
OTA update for TS0601_thermostat deactivated for the time being.

Problem clarification.
https://github.com/Koenkk/zigbee2mqtt/issues/7941#issuecomment-906243509

closed 2021-11-01 by github-actions[bot]