Koenkk/zigbee2mqtt #8803: Can only pair via specific routers

nickw444, 2021-09-19:
## What happened

Can only pair end devices via specific routers on the network

## What did you expect to happen

Can pair end devices via any routers on the network

## How to reproduce it (minimal and precise)

1. Pair a number of router devices (>5 routers) to the network. 
2. For each router, enable pairing for it only, attempt to pair an end device.
3. For some routers pairing will not take place, even with multiple attempts. For other routers pairing works almost immediately. 

## Debug info
**Zigbee2MQTT version:** 1.21.1-1 (commit: unknown) in hassio supervisor, running on Rpi4 4GB
**Adapter hardware:** CC1352P-2 (based on RF-Star CC2652P2)
**Adapter firmware version:** zStack3x0, 20210708

## Further details

Initially I found https://github.com/Koenkk/zigbee2mqtt/issues/7762; which I believed was the root cause of my problem. I had an older network originally running from a CC2530 based coordinator.

However, I have completely nuked my existing network, including re-flashing my CC2652P2 stick, nuking all z2m data (including coordinator backup), and re-paired all existing devices multiple times and still appear to have this same issue.

It does not appear to matter what channel, Pan ID, or network key is used.

I tried reproducing initially on a smaller test network (2 repeaters - 1x Tradfri repeater, 1x DIY CC2530, 2 end devices, CC2652P2 based coordinator), but couldn't reproduce the problem in this configuration. 

Instead have needed to re-create my entire production network (including repairing some annoying to access end devices). The issue seems to present itself as the network gains complexity. After adding 5 router devices, it appears to become a problem. End devices will only pair through certain routers. You can see this in the map:

[image]

I did some further digging this weekend using a CC2531 with sniffer firmware + wireshark. I have captured the following traces:

[Archive.zip](https://github.com/Koenkk/zigbee2mqtt/files/7191011/Archive.zip)

(yes I know I've exposed my network key, I will roll it when/if this reaches a resolution)

To explain these traces, there are 5 routers on the network;

|IEEE Address| Network Address | Model | Nickname |
|--|--|--|--|
|0xec1bbdfffe9cf5a2 |0xE63A| IKEA LED1732G11 | Nicks Bedside |
|0x588e81fffe021892 |0x6CFA| IKEA E1746 | Living Room Repeater |
|0x588e81fffe0205b1 |0x049F| IKEA E1746 | Lounge Repeater |
|0x00124b001fc64014 |0xF82B| Custom devices (DiY) CC2530.ROUTER |Living Room Noise Repeater|
|0x00124b000add55c6 |0xE671| Custom devices (DiY) CC2530.ROUTER |Study Repeater|

### `failed-join-bathroom-motion-via-study-lounge.pcapng`

Attempt to join "Bathroom Motion" (`0x00158d00063372bf (0x5FB2)`) via "Study Repeater" (`0x00124b000add55c6 (0xE671)`).

Successfully joins via "Lounge Repeater" (`0x588e81fffe0205b1 (0x049F)`)

### `failed-join-bathroom-via-study-lounge-bedside.pcapng`

Attempt to join "Bathroom Door" (`0x00158d000451d91a (0x8DFC)`) via "Study Repeater" (`0x00124b000add55c6 (0xE671)`).

Attempt to join via "Lounge Repeater" (`0x588e81fffe0205b1 (0x049F)`)

Successfully joins via "Nicks Bedside" (`0xec1bbdfffe9cf5a2 (0xE63A)`)

### `failed-join-front-door-button-via-living-room-noise-living-room-lounge.pcapng`

Attempt to join "Front Door Button" (`0x00158d00045075e9 (0xFD40)`) via "Living Room Noise Repeater" (`0x00124b001fc64014 (0xF82B)`)

Attempt to join via "Living Room Repeater" (`0x588e81fffe021892 (0x6CFA)`)

Successfully joins via "Lounge Repeater" (`0x588e81fffe0205b1 (0x049F)`)

### `failed-join-lemon-tree-via-bedside-via-louge.pcapng`

Attempt to join "Lemon Tree" (`0x00124b000611b998 (0x997B)`) via "Nicks Bedside" (`0xec1bbdfffe9cf5a2 (0xE63A)`)

Successfully joins via "Lounge Repeater" (`0x588e81fffe0205b1 (0x049F)`)

### `failed-join-living room-button-via-living-room-bedside-lounge.pcapng`

Attempt to join "Living Room Button" (`0x00158d00028f5873 (0x3B45)`) via "Living Room Repeater" (`0x588e81fffe021892 (0x6CFA)`). 

Attempt to join via "Nicks Bedside" (`0xec1bbdfffe9cf5a2 (0xE63A)`)

Successfully joins via "Lounge Repeater" (`0x588e81fffe0205b1 (0x049F)`)

### `failed-join-living-room-motion-via-many-final-through-nicks-bedside.pcapng`

Attempt to join "Living Room Motion" (`0x00158d000421a072 (0x5B83)`) via many different repeaters. Finally succeeds via "Nicks Bedside" (`0xec1bbdfffe9cf5a2 (0xE63A)`)

Koenkk, 2021-09-20:
Thanks for the well detailed problem description and sniffs!

I can provide a firmware with additional debug logging, I've checked `failed-join-bathroom-motion-via-study-lounge.pcapng` and I see the problem is on the coordinator side since the `Update device` is not followed up by a `Transport Key`. But before doing this, can you confirm that you did NOT reflash your coordinator after repairing all devices?

nickw444, 2021-09-20:
> But before doing this, can you confirm that you did NOT reflash your coordinator after repairing all devices?

Yep the coordinator was reflashed first along with nuking the z2m directory and setting up with a new Pan ID and network key; then I began re-pairing all devices onto this new network.

Koenkk, 2021-09-21:
Can you provide the herdsman debug logging when trying to pair a device through a router fails?

I assumed you are currently running `CC1352P2_CC2652P_launchpad_coordinator_20210708.zip`, if not **don't** flash this fw.

[znp_CC1352P_2_LAUNCHXL_tirtos_ccs_launchpad.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/7205754/znp_CC1352P_2_LAUNCHXL_tirtos_ccs_launchpad.hex.zip)

nickw444, 2021-09-25:
Hey Koen, I flashed the debug firmware to my stick, enabled debug and had a crack re-pairing one of my devices (`0x00158d000421a072 (0x2582)`) via one of the problematic routers;

Interestingly no debug messages appear from the coordinator when attempting pairing through the malfunctioning router in any of the captures below. Pairing via a working router appears to work/show debug messages.

You'll see these in the attached zip file, which contains the `pcap` and the log from z2m (including herdsman debug log)

The two devices I've been playing with have the following IEEE addresses:

* `0x00158d0004507670`
* `0x00158d000421a072`

## Capture + Log 1

[z2m-debug-1.zip](https://github.com/Koenkk/zigbee2mqtt/files/7228942/z2m-debug-1.zip)

## Capture + Log 2

In this capture, I tried pairing via a few different routers, until the last succeeded

[z2m-debug-2.zip](https://github.com/Koenkk/zigbee2mqtt/files/7228940/z2m-debug-2.zip)

## Capture + Log 3

[z2m-debug-3.zip](https://github.com/Koenkk/zigbee2mqtt/files/7228939/z2m-debug-3.zip)

This one is probably the most interesting of the lot; 

* I accidentally unpaired one of the malfunctioning routers (`0x00124b001fc64014`).
* Attempted to re-pair via Lounge Repeater (`0x588e81fffe0205b1 (0x049F)`) without success. No messages written in herdsman log AFAICT.
* Re-pair successful when Permit Join (All) enabled. Looks like key transfer was direct from coordinator. 
* Permit join via `0x00124b001fc64014 (0xC9A5)`
* Attempt to pair 0x00158d0004507670 (0x2EA9) and 0x00158d000421a072 (0xB584). Pairing does not proceed.

Koenkk, 2021-09-25:
@nickw444 what does `z2m-debug-1.zip` contain? I need a log when pairing fails (and no other attempts in it )

nickw444, 2021-09-26:
All of those contain both the debug log from the z2m container (supervised in hassio) as well as a corresponding packet dump. (followed the steps [here](https://www.zigbee2mqtt.io/information/debug.html#home-assistant-ossupervised-addon) under the **Home Assistant OS/Supervised addon** heading)

As mentioned above, no debug messages appear from the coordinator when attempting pairing through the malfunctioning router, which I imagine is what you're seeing here - a lack of debug messages. I experienced this for all of the captures above, only pairing via a functioning router appeared to show logging.

Koenkk, 2021-09-27:
@nickw444 but what did you try in "Capture + Log 1", you explained it for the others but not for "Capture + Log 1", (e.g. for "Capture + Log 2"):

> In this capture, I tried pairing via a few different routers, until the last succeeded

What I need is a capture + log which only contains a failing attempt to pair a device via a router. (to prevent log pollution)

nickw444, 2021-09-28:
Apologies, I think I misinterpreted your first message. I vaguely recall that the first capture was toggling pairing across a few different routers until it finally paired, similar to the last capture. 

I've done a new capture, this time attempting to pair a single device to a single router (`0x00124b001fc64014 (0xE516)`).

[Archive.zip](https://github.com/Koenkk/zigbee2mqtt/files/7247743/Archive.zip)

I've taken a screen recording of the process too, in case that helps? You can see in this video the beacon request and response, but no activity in the z2m log.

https://youtu.be/gYGJKxjHdSc

Koenkk, 2021-09-29:
Exactly what I needed (sniff + logs)

Can you again provide a herdsman debug log + sniff of just 1 pairing attempt with the fw below:
[znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/7254897/znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip)

Note, pairing might always succeed now, let me know if this is the case.

nickw444, 2021-09-30:
Awesome, I've just flashed this firmware, and attempted to pair the same device (0x00158d0004507670) again to the same router (`0x00124b001fc64014 (0xE516)`).

Here's the pcap and the herdsman debug log
[Archive 2.zip](https://github.com/Koenkk/zigbee2mqtt/files/7256270/Archive.2.zip)
:

Didn't successfully pair, and didn't see any new log lines emitted whilst attempting to pair the device.

Koenkk, 2021-09-30:
@nickw444 to be sure, can you try repairing the router that you are trying to pair via (I see the joining procedure does not start at all while it did in https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-929687244)

nickw444, 2021-10-01:
@Koenkk I've re-paired the router in question (using Permit Join via All), it looks as though to have the coordinator as its parent on the map. 

I've done another capture of pairing the same Aqara button `0x00158d0004507670` to the same router `0x00124b001fc64014`, but still didn't manage to successfully
[Archive 3.zip](https://github.com/Koenkk/zigbee2mqtt/files/7265166/Archive.3.zip)
 pair. Hopefully this initiated the pairing procedure in this capture.

If not, I can try again, is there anything I need to look for in Wireshark to confirm whether it had begun? Just to save the back and forth here.

Koenkk, 2021-10-02:
@nickw444

- What device is `0x00124b001fc64014`?
- Could you share your `data/coordinator_backup.json`?

nickw444, 2021-10-02:
> What device is 0x00124b001fc64014?

This is a "Custom devices (DiY) CC2530.ROUTER" (Just a CC2530 + 2591 PA dev board)

> Could you share your data/coordinator_backup.json?

[coordinator_backup.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/7272299/coordinator_backup.json.zip)

Interestingly, the coordinator backup doesn't have that router IEEE address in it, but I am not sure if that was due to an unclean shutdown of my Pi. (Currently in the process of changing OS, so can't boot it up at this very moment, this was taken from an FS snapshot just now)

Should this backup also list all devices on the network? It only has a small amount. There are over 20 devices paired on this network.

Koenkk, 2021-10-03:
When repairing the CC2530.ROUTER, did you first remove it (non a force remove)? if not, do this, try to pair the device via the router and provide the log + sniff.

nickw444, 2021-10-03:
Hey Koen, yep, this was done by removing it from Z2M (and a toast for "device left network" appeared).

In the mean time, I've tried to reproduce this away from my Pi on my desktop. I've paired a minimal network which I am able to reproduce with:

- 2x IKEA Tradfri Repeaters (E1746)
- 1x CC2530 Custom Devices Repeater

I've provided in the attached zip:

| filename | description |
|--|--|
| `test-network-rebuild-failed-join-cc2530.log` | Log file produced by z2m (incl. herdsman-debug) whilst building the test network.  |
| `test-network-rebuild-failed-join-cc2530.pcapng` | pcapng captured whilst building the test network. Includes some failure to join via CC2530 router during rebuild |
| `test-network-failed-join-only.pcapng` | pcapng captured whilst pairing a single device to the CC2530 router after rebuild. |
| `test-network-failed-join-only.log`| pcapng captured whilst pairing a single device to the CC2530 router after rebuild. |
| `data` | z2m data directory for the test network |

[failure-to-join-debug.zip](https://github.com/Koenkk/zigbee2mqtt/files/7274731/failure-to-join-debug.zip)

**Devices List**
[image]

**Network Map**
[image]

## Sequence of Events

### Start Rebuilding

Koenkk, 2021-10-04:
Still no debugging appeared, now that we know for sure it has nothing to do with the backup mechanism, this must be a bug in the firmware. I was checking https://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/5.20.00.52/exports/changelog.html and found:

>  When TC settings do not mandate TCLK update, TC does not process APS packets w/ well-known TCLK encryption for remote multi-hop devices

Looks likes TI might already have fixed this bug, however I didn't update to 5.20 yet since it gave issues (current fw is based on 5.10). I expect the next release soon (last was released in July and they release every quarter), will ping you when I have a fw ready.

nickw444, 2021-10-04:
Ok sounds good. 

I might try an older firmware release in the interim to see if that improves the situation (however IIRC one of the bugs in the previous release was inability to pair to a specific router so might be hard to know for sure if this is the case)

hnykda, 2021-11-10:
Be a nice bot and don't close it.

hnykda, 2021-11-11:
For anyone interested, flashing January firmware might likely get you moving in the interim. As per this discord discussion:

[image]

Koenkk, 2021-11-14:
New firmware is live, please check with the following fw: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

realalexandergeorgiev, 2021-11-15:
Tried the new firmware. Didn't change anything for me. Can't add new devices. Thank you for trying ♥️

hnykda, 2021-11-15:
Same here. Anyway, there was simply something wrong when I switched to slaesh - I tried different firmwares from January, July and the one now, but none of them could connect new devices. I deleted everything, changed pan_id and network_key and repaired everything again flawlessly :shrug: ...

nickw444, 2021-11-15:
> I deleted everything, changed pan_id and network_key and repaired everything again flawlessly 🤷 ...

It'll be interesting to see if this continues to be the case. I found in my testing the problem got worse after a few days after the initial setup.

realalexandergeorgiev, 2021-11-16:
I was playing around today and noticed two things that might be interesting:

1. I was not able to add the IKEA on/off switch, but the Xiaomi wireless single button WXKG11LM. I was only able to do this in the basement, but I can repeat removing and adding it. It works. But it doesn't anywhere else in the house. Maybe a bug? Interferences??
2. I noticed a thing when moving from CC2531 to CC2652 - some buttons had really bad LQI besides them being near a router like light. Was not the case before. I thought the network just needs time. Anyway. I took my fresh paired button, kept pressing the button and walked inside the house. The LQI went to 3 and at some point it was not able to send click events any more. Besides being near many routers. Something seems wrong with routing. It feels like it can only send to the coordinator.

hnykda, 2021-11-18:
@nickw444 You were right, I now can't add new TRV valves of the same type that I could add at the beginning when I restarted everything (and when I used cc2531) :cry: . Now it has been just 3 days... Is there a chance that when I use older firmware version, it will work?

Koenkk, 2021-11-19:
To all the people still experiencing this issue; did you all migrate from a CC2531 on the 1.2 firmware to CC2652/CC1352 **without** starting from scratch?

realalexandergeorgiev, 2021-11-19:
I did not start from scratch, correct. I can check the exact firmware later, still have the CC2531

nickw444, 2021-11-19:
Neither, in every instance I have started fresh - In fact, when I set up my CC2652 I missed the release note to change Z2M repo in Hass IO, so didn't even have the capability to do the migration 😆

hnykda, 2021-11-19:
I migrated from CC2531, firmware version `2021127` after massive network outages and issues.

Then I got the Slaesh stick with the June firmware, migration went fine (i.e. the network was up), but I couldn't join new devices (and some of them simply "dropped off" during the migration, specifically one IKEA Fyrtur blind - and I couldn't reconnect it). See #9565 .

So I [started fresh](https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-969190346) and was happy that I can pair stuff again (including that blind, for example), but now I cannot pair *some of my* [devices again](https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-973196821). I can pair an IKEA repeater, but not TRVs (that worked fine when I migrated from CC2531 co Slaesh).

Koenkk, 2021-11-19:
@hnykda could you:
- try to pair a device and make a sniff of this.
- observe what router send the `Association Response` to the device you are trying to pair. This is the router the device is trying to pair via
- see that the router sends an `Update Device` to the coordinator, observe that the coordinator does NOT send a `Transport Key` back (this is the problem)
- Check if the router is in the `devices` list of `data/coordinator_backup.json`, if yes, also check if it has a `link_key` in it. Example:
[image]
- Now repair the router close to the coordinator
- Stop zigbee2mqtt, check if the router is in `data/coordinator_backup.json` now with a `link_key`.
- If yes, put the router back to its original location and try to repair your device

hnykda, 2021-11-20:
Hi. I probably somehow failed in doing it... Even when I moved the TRVs next to the coordinator, I cannot connect it at all and I don't see anything that would point to the TRVs actually sending anything. Maybe the TRVs just magically broke? Anyway, I tried with an IKEA blinds controller. I couldn't pair it when being far from coordinator, but I could when being in the coordinator's range. Unfortunately, when I am not close to the coordinator, I don't see any "Association Response" (with the button nor the TRVs - with those though, I don't see it even when close to it). Therefore I didn't even get to the second bullet point. 

Also, I suspect I am somehow haunted - when trying to pair one of the 3 TRVs that don't work now, even though I haven't seen anything in wiresharks logs (but I could've just stopped the capture as I was trying it like 50 times), it got paired. The other one didn't at all even when putting it next to coordinator (it's the `_TZE200_zuhszj9s` / `TS0601`, so the pairing is usually trivial). I suspect it "caughted" a glimpse of the coordinator signal somehow?

All the logs here: https://easyupload.io/m/0wrw4t

Koenkk, 2021-11-20:
- unsuccessful_pairing_of_trv*: something strange going on here, device does seems to send beacon requests or beacon request are not answered by routers.
- failed_button_far: this one shows the problem as I explained in https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-974306405 , you see that the update device on 200 is not followed up by a transport key from the coordinator. Could you continue debugging this from step 4 of https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-974306405 ?

hnykda, 2021-11-21:
OK, so I gave it a try. Based on the sniff packet 200 I sent you earlier when trying to pair the button, I found out that the source router for that pairing was my IKEA repeater paired earlier with network address `cbca`. I went to `coordinator_backup.json` and haven't found any device with `nwk_address: "cbca"`:
[image]

Should I now remove that IKEA repeater and then pairing it again or just pairing it again without removing first (is that even possible)?

Koenkk, 2021-11-21:
- to be sure, also check if the ieee address it not in the coordinator_backup.json, using the z2m frontend you should be able to find the ieee address of the device
- if no, then repair the device close to the coordinator, stop z2m and check if it is in coordinator_backup.json
- if it is in coordinator_backup.json now, move the router back to its original location and try to pair the button again

realalexandergeorgiev, 2021-11-21:
Should one see all routers there? (Even if not starting from scratch?) I see only 2 devices there. The two buttons I recently paired in the basement 5days ago (see above).

francisp2, 2021-11-22:
> Should one see all routers there?

Good question. I have 31 Zigbee devices (excluding the coordinator), but only 23 devices in coordinator_backup.json

Koenkk, 2021-11-22:
> Should one see all routers there? (Even if not starting from scratch?) I see only 2 devices there. The two Xiaomi buttons(!) I recently paired in the basement 5days ago (see above).

@castorw does the coordinator_backup.json contain all the router or just the ones with a specific link key?

castorw, 2021-11-22:
@Koenkk All devices known to the coordinator, including those connected via routers should be backed up. Although I am not entirely sure ZNP stores these devices in `ADDRMGR` if they do not exchange APS keys.

Can any of you guys having less devices in `coordinator_backup.json` than actually paired in the network provide me with the `coordinator_backup.json` and NV dump from the coordinator? I could investigate further in the matter.

To extract NV contents you can use: https://github.com/zigpy/zigpy-znp/blob/dev/zigpy_znp/tools/nvram_read.py.

hnykda, 2021-11-22:
@Koenkk 

> to be sure, also check if the ieee address it not in the coordinator_backup.json, using the z2m frontend you should be able to find the ieee address of the device

OK, I actually found that under the IEEE address (but different network address) :shrug: : 
```
    {
      "nwk_address": "76da",
      "ieee_address": "84fd27fffe702988",
      "is_child": false,
      "link_key": {
        "key": "507de2a6bfea12c60f3a2505518437db",
        "rx_counter": 0,
        "tx_counter": 5275
      }
    },
```

now I don't know what you want me to do, because you wrote "if no, then ..." 

What should I do "if yes" :grimacing: ?

And I seem to have the devices there.

castorw, 2021-11-22:
@hnykda The network address is not as important as the IEEE address. The network address may change. However, what is interesting is that APS key `rx_counter` is 0.

* What kind of device that is again?
* Could you please check the network map to see if the device is connected directly to coordinator?

hnykda, 2021-11-22:
OK, didn't know that, thanks.

This is the router: https://www.zigbee2mqtt.io/devices/E1746.html#ikea-e1746 . I had no troubles of pairing this device even when I was far from the coordinator, but it's maybe because my house is so small that this device can simply reach it as any other router in my house according to the network map:
[image]

The red circle is the IKEA repeater.

castorw, 2021-11-22:
@hnykda :
* The problem is that you can't pair devices to certain routers right?
* Is the problem present after the router has been freshly paired to network or after restoring from backup?
* If after restoring the backup, did the problem exist with the original network formation?

hnykda, 2021-11-22:
Eh, I think I actually have two problems:

1. TRVs not pairing at all after a while even though the same type paired fine [right after I simply started from scratch](https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-974038026)
2. Not being able to pair devices unless pairing them close to the coordinator (which in my case isn't such a huge problem, as my house is fairly small, but still annoying and want to provide the feedback - once again, thanks a lot for doing all of this)

> Is the problem present after the router has been freshly paired to network or after restoring from backup?
I believe this problem wasn't present after a fresh network (at least I haven't recognized that), so there wasn't any backup as far as I can tell (I simply deleted the old zigbee2mqtt data directory and started from scratch with a new network channel) :thinking: .

castorw, 2021-11-22:
@hnykda Could you contact me on telegram? @castorko
I am pretty curious about the state of your network and there are indications of possible problems with ZNP adapter handling in zigbee-herdsman code.

francisp2, 2021-11-23:
@castorw 
output of /nvram_read.py :
[backup.txt](https://github.com/Koenkk/zigbee2mqtt/files/7589384/backup.txt)
 coordinator_backup.json : 
[coordinator_backup_json.txt](https://github.com/Koenkk/zigbee2mqtt/files/7589399/coordinator_backup_json.txt)
database.db: 
[database_db.txt](https://github.com/Koenkk/zigbee2mqtt/files/7589408/database_db.txt)

jcastro, 2021-12-02:
I'm now able to pair with new devices using the latest version of zigbee2mqtt (1.22.1-1) with latest firmware for my device (Zzh! CC2652R Multiprotocol RF Stick)

castorw, 2021-12-05:
@francisp2 What exactly? Cannot pair certain devices? Cannot pair devices at all? Cannot pair through certain routers?
Are you trying with the latest Z2M version?

francisp2, 2021-12-06:
@castorw Can't pair devices through certain routers, and can't pair unless I just restarted Zigbee2mqtt. I'm using 1.22.1 with a Tube Ethernet coordinator.

Koenkk, 2021-12-07:
TI released another SDK update and it looks they added more joining fixes, for those still experiencing this issue please try with `20211207`: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

hnykda, 2021-12-08:
I wanted to try it, but first I wanted to replicate [the issue I had in here](https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-975928557) but I can't replicate it - the device pairs just fine now (not the TRVs, but these are probably somehow broken). Starting to drive me crazy :joy: .

jcastro, 2021-12-08:
Using the latest version of zigbee2mqtt (1.22.1-1) with latest firmware (
Z-Stack 3.x.0 20211207) for my device (Zzh! CC2652R Multiprotocol RF Stick) and I can still pair new devices. Just wanted to do an update on it.

nickw444, 2021-12-13:
Have upgraded my coordinator to `20211207` over the weekend, and re-paired routers. Looks like the issue is resolved (from what I can tell), managed to pair via a router. Will see if this stays working for the next few days.

realalexandergeorgiev, 2021-12-13:
I updated the firmware. Paired a button, worked, moved it inside house, lost connection 10 minutes ago. Unable to find the network. The room it is in has 3 routers. I kept pressing the button while moving there. 

Something is broken with routing somehow. I will go back to CC2531 once I find the time. Then I can tell if its related to the chipset (firmware).

nickw444, 2021-12-14:
Dang! Just tried pairing a new Aqara button away from the coordinator but unfortunately it seems to not be working again 🤨

francisp2, 2022-01-14:
I still don't know if it is normal that  coordinator_backup.json contains less devices then configuration.yaml and database.db.

nickw444, 2022-01-17:
Agreed, still have some funny things going on in my network, particularly with some devices dropping off the network a few days after being re-paired.

RaveGun, 2022-01-19:
Switched from CC2531 to Slaesh's stick on Z2M Edge.
Just the sticks switched and the new serial path configured.
There network worked fine, no issues.
Some weeks later tried to add a new light bulb. Although is supported it paired as an unknown device. I deleted it and tried again. It did not worked at all anymore.
Later I added the light bulb to the ZHA network that I have also running with a Deconz stick, no issues.

Several days later tried to pair a Phillips Hue wall switch. Nothing. Just now I tried to add a TVR. Nothing. It is like the network does not switch into pairing mode.

Both just paired fine to the ZHA.

I really wanted to have the Z2M as the main network and the ZHA for testing purposes but I keep adding more and more devices there.

If I start from scratch, not so keen, are somehow the names of the devices kept so that I don't break anything else in home assistant?

Thank you

castorw, 2022-01-19:
I believe this is related to #9117. I will be investigating further into this issue as per https://github.com/Koenkk/zigbee2mqtt/issues/9117#issuecomment-1016890665.

hnykda, 2022-01-19:
I don't want to derail the conversation, but want to follow up on this in case it's relevant (but it might not!).

Basically, I managed to somehow pair those few missing TRVs by pairing them close to coordinator. It just worked after trying it like 20 times... More about it here: https://github.com/Koenkk/zigbee2mqtt/discussions/10774 . What surprised me a bit was that they got the same name, but that could be because I think I copied friendly name somewhere in the z2m data folder. So that's my story about successful pairings :shrug: .

Second one is weirder. I accidentally hard-switched off (like physically) one of my routers (IKEA Floalt) for a few hours and the whole network started to get crazy. Many devices dropped off the network, even though they were closer to the coordinator or were routers on their own, like other IKEA Floalt. My network is heavily "over centered", almost every device would reach the coordinator and routers wouldn't be needed to extend the range in my case. Z2M simply showed "No route to network (205)". When I (hard)restarted these devices (by fuses down/up), the network works fine now again. But it's pretty scary for me, definitely not what I would expect from mesh.

[1 more reply]

closed 2022-05-25 by github-actions[bot]