Koenkk/zigbee2mqtt #8805: TuYa TS0202 motion sensor no battery + tamper

dcboy87, 2021-09-19:
## What happened

I paired [this ](https://www.aliexpress.com/item/1005003036701978.html) motion sensor, everything works so far, exept tamper and Battery level. It seems, this device doesn't support both values. Also in frontend, where power source is indicated, theere is only a flashing red % symbol

[image]

## What did you expect to happen
- showing right power source
- working tamper  or remove tamper
- correct battery level or remove battery level

## How to reproduce it (minimal and precise)
pair and see
## Debug info

Zigbee2MQTT version
    1.21.1-dev commit: 27b36b31

Coordinator type
    zStack30x

Coordinator revision
    20190425

Frontend version
    0.6.7

adriannowaq, 2021-09-22:
You need to write your own converter for this

adriannowaq, 2021-09-23:
You should try these:
[HOW TO SUPPORT NEW DEVICES](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
)

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0202'], 
    model: 'TS0202', 
    vendor: 'undefined',
    description: 'motion sensor', 
    fromZigbee: [fz.ias_contact_alarm_1_report, fz.battery], 
    toZigbee: [], 
    exposes: [e.battery(), e.occupancy(), e.tamper(), e.linkquality() ], 
};

module.exports = definition;
```

closed 2021-11-01 by github-actions[bot]