Koenkk/zigbee2mqtt #8808: QS-Zigbee-D02-TRIAC-2C-LN - transition, brigtness_move or brightness_step not working

wroadd, 2021-09-19:
## What happened
"transition" and "briteness_step" not working with MQTT command and also with config parameter with QS-Zigbee-D02-TRIAC-2C-LN (same command works with IKEA color bulb 600 lm)
No any sign of error in the logs.

## What did you expect to happen
I used https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-2C-LN.html to have transition.
Have a transition effect
## How to reproduce it (minimal and precise)
- I've defined the "transition" parameter in devices.yaml
- I've called an mqtt publish command on the proper channel on HA
service: mqtt.publish
data:
  topic: zigbee2mqtt/Nappali dimmer (Dim001)/l1/set
  payload: '{ "brightness_step":40"}' 
and  payload: '{ "transition":5, "brightness":254 }' 
payload: '{ "brightness_move":40"}' 

## Debug info
Zigbee2MQTT version: zigbee2mqtt@1.21.1
Adapter hardware: CC2538 + CC2592 - Zigbee dongle - ZB3.0
Adapter firmware version: 
Coordinator type zStack30x
Coordinator revision 20200327

Herdsman log: 
Zigbee2MQTT:debug 2021-09-19 19:52:36: Publishing 'set' 'brightness_step' to 'Nappali dimmer (Dim001)'
2021-09-19T17:52:36.253Z zigbee-herdsman:controller:endpoint Command 0x5c0272fffe16b181/1 genLevelCtrl.step({"stepmode":0,"stepsize":5,"transtime":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-09-19T17:52:36.253Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x5c0272fffe16b181:40180/1 (0,0,1)
2021-09-19T17:52:36.254Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":40180,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":47,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[1,22,2,0,5,0,0]}}
2021-09-19T17:52:36.255Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,244,156,1,1,8,0,47,0,30,7,1,22,2,0,5,0,0,114]
2021-09-19T17:52:36.257Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,244,156,1,1,0,78,0,217,11,191,0,0,7,24,82,10,0,0,16,1,244,156,29,176]
2021-09-19T17:52:36.257Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,244,156,1,1,0,78,0,217,11,191,0,0,7,24,82,10,0,0,16,1,244,156,29,176]
2021-09-19T17:52:36.258Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,244,156,1,1,0,78,0,217,11,191,0,0,7,24,82,10,0,0,16,1,244,156,29] - 176
2021-09-19T17:52:36.258Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":40180,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":78,"securityuse":0,"timestamp":12520409,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,82,10,0,0,16,1]}}
2021-09-19T17:52:36.260Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":82,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":40180,"endpoint":1,"linkquality":78,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2021-09-19 19:52:36: Received Zigbee message from 'Nappali dimmer (Dim001)', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
2021-09-19T17:52:36.265Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2021-09-19 19:52:36: MQTT publish: topic 'zigbee2mqtt/Nappali dimmer (Dim001)', payload '{"brightness_l1":254,"brightness_l2":252,"linkquality":78,"state":"ON","state_l1":"ON","state_l2":"ON"}'
Zigbee2MQTT:info  2021-09-19 19:52:36: MQTT publish: topic 'zigbee2mqtt/Nappali dimmer (Dim001)/l1', payload '{"brightness":254,"state":"ON"}'
Zigbee2MQTT:info  2021-09-19 19:52:36: MQTT publish: topic 'zigbee2mqtt/Nappali dimmer (Dim001)/l2', payload '{"brightness":252,"state":"ON"}'
2021-09-19T17:52:36.282Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,1,47,233,254,25,68,129,0,0,8,0,244,156,1,1,0,70,0,19,12,191,0,0,5,8,22,11,2,0,244,156,29,61]
2021-09-19T17:52:36.283Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,47,233,254,25,68,129,0,0,8,0,244,156,1,1,0,70,0,19,12,191,0,0,5,8,22,11,2,0,244,156,29,61]
2021-09-19T17:52:36.283Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-09-19T17:52:36.284Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-09-19T17:52:36.284Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,47,233,254,25,68,129,0,0,8,0,244,156,1,1,0,70,0,19,12,191,0,0,5,8,22,11,2,0,244,156,29,61]
2021-09-19T17:52:36.284Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,47] - 233
2021-09-19T17:52:36.284Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":47}
2021-09-19T17:52:36.285Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,8,0,244,156,1,1,0,70,0,19,12,191,0,0,5,8,22,11,2,0,244,156,29,61]
2021-09-19T17:52:36.285Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,8,0,244,156,1,1,0,70,0,19,12,191,0,0,5,8,22,11,2,0,244,156,29] - 61
2021-09-19T17:52:36.285Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":40180,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":70,"securityuse":0,"timestamp":12520467,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,22,11,2,0]}}
2021-09-19T17:52:36.286Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":22,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":2,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":40180,"endpoint":1,"linkquality":70,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-09-19T17:52:36.286Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2021-09-19 19:52:36: MQTT publish: topic 'zigbee2mqtt/Nappali dimmer (Dim001)', payload '{"brightness_l1":254,"brightness_l2":252,"linkquality":70,"state":"ON","state_l1":"ON","state_l2":"ON"}'
Zigbee2MQTT:info  2021-09-19 19:52:36: MQTT publish: topic 'zigbee2mqtt/Nappali dimmer (Dim001)/l1', payload '{"brightness":254,"state":"ON"}'
Zigbee2MQTT:info  2021-09-19 19:52:36: MQTT publish: topic 'zigbee2mqtt/Nappali dimmer (Dim001)/l2', payload '{"brightness":252,"state":"ON"}'

RMNB, 2021-09-19:
I also can't get brightness_move to work with IKEA LED1732G11.

iamhyde, 2021-09-20:
same one, the dimming does not work，only on/off
#8575

karaminder, 2023-01-13:
I still have the same issue today. Feels like it's stuck on 50% brightness but no amount of adjustment in Z2M affects the light output.

ONLY On/Off work.