Koenkk/zigbee2mqtt #8810: RWL022 Re-enable OTA

JoachimEngelmann, 2021-09-19:
At the implementation of RWL022 OTA was disabled, because it caused some OTA errors (https://github.com/Koenkk/zigbee2mqtt/issues/6359). 
I think this was caused because there was no OTA software available at this time, but some days ago Philips puplished a OTA update for the RWL022 (Dimmer switch v2). So it should be tested again and enabled if it works

Bascht74, 2021-11-20:
@JoachimEngelmann Hi, I got the new dimmer (RWL022) as well and if I try to update the dimmer I got this in the log: 

`No image available for imageType '281'`

I am using the latest dev...

Did you successfully update your RWL022

Bascht74, 2021-11-20:
it seems to be the same error here: https://github.com/Koenkk/zigbee2mqtt/issues/6359#issuecomment-785258081
That is why @Koenkk disabled OTA for this device. 
But now it seems to be enabled again: 
https://github.com/Koenkk/zigbee2mqtt/issues/9257

Bascht74, 2021-11-20:
Seems that nobody found the URL of the firmware file so far...
Is there any info about sniffing the URL when updating via the hue bridge?

JoachimEngelmann, 2021-11-21:
In this thread someone sniffed a hue update package (https://github.com/dresden-elektronik/deconz-ota-plugin/issues/10). It seems the hue bridge downloads the update and then pushes it to the device. So if you sniff all network traffic of the bridge it should be possible to get the firmware file.
Sadly i don't have a hue bridge, otherwise i would have tried it already