Koenkk/zigbee2mqtt #8813: (Dashboard display) Issues with Siterwell GS361A-H04

ovizii, 2021-09-20:
I connected a new Siterwell GS361A-H04 which is supposed to be supported according to: https://www.zigbee2mqtt.io/devices/GS361A-H04.html#siterwell-gs361a-h04 

It is now showing up in zigbee2mqtt but I can see a few quirks and am unsure if this is a bug or if I can solve it myself via some magic configuration.

z2mqtt dashboard, 3x state doesn't really help. 
Why is running state shown as N/A?
[image]

The info seems to be there just not displayed in the dashboard
[image]

[image]

## Debug info
Zigbee2MQTT version: 1.21.1-1
Adapter hardware: cc2652rb 
Adapter firmware version: not sure where to find it but I am using a recommended adapter: Slaesh's CC2652RB stick

datag, 2021-10-21:
Here too:

Sending `{"current_heating_setpoint": ""}` to the `*/get` topic results in
`Zigbee2MQTT:error 2021-10-21 22:40:05: No converter available for 'get' 'current_heating_setpoint' ()`

manufacturerName of my device: `_TZE200_jeaxp72v`

Also running latest dev branch. Willing to debug but might need some hints.

closed 2021-11-28 by github-actions[bot]

jufo84, 2022-01-23:
Same here both values shows null

Andy-Hennah, 2022-06-16:
[image]

I'm seeing the the thing.  I'm happy to provide logs etc. (but I might need a little guidance).  This is on z2m version 1.25.2.

Andy-Hennah, 2022-06-16:
This is what the state looks like (from the z2m interface)
[image]

Woodpeckercz, 2022-08-03:
Same here, I have two of these and both show position n/a and running state null

toby1984, 2022-09-28:
Same here, anyone know how to debug this ?

Dark-Gizmoa, 2023-01-04:
exact the same problem :-(

GamerClassN7, 2023-03-06:
Hame problem here any news ?

josephfh, 2023-04-14:
Same problem here. I can provide logs if anyone can help

GamerClassN7, 2023-04-16:
> Same problem here. I can provide logs if anyone can help

Unforchunatelly, noone is responding , do you have same probles foloved with problem of sending manualy set values to home assitant ? as well ?  thanks for answer

GamerClassN7, 2023-04-16:
https://github.com/Koenkk/zigbee2mqtt/issues/17365 I created new  Issue dedicated to reading data from device :) hopefully it gets some notice :)

Setka20, 2023-10-19:
I have the same problem.

massenmacher, 2023-10-23:
For me also still a problem :/

veeroohre, 2024-11-15:
why is this issue closed?

Soulfly999, 2025-05-06:
The Problem still exists! Zigbee2MQTT-Version [2.3.0]