Koenkk/zigbee2mqtt #8815: All devices become unavailable

Jens-Wymeersch, 2021-09-20:
## What happened

Since I added the last 20 devices (by sunsetting my HUE bridge and ZHA environment), Z2M has been quite unstable. I had to restart every other day. I don't really understand why. But all devices become unavailable. 

## What did you expect to happen

I expected no major problems. 

## How to reproduce it (minimal and precise)
My environment is a standalone RPI 3B+ running Home Assistant (almost solely setup for Z2M)

All devices have the same log error : 

`Zigbee2MQTT:error 2021-09-18 21:05:38: Publish 'set' 'state' to 'Lamp Colour L1 Master' failed: 'Error: Command 0x00158d0004774bd0/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'`

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652R
Adapter firmware version: HA version 1.21.1-1

Koenkk, 2021-09-20:
- What firmware version are you using? If not using the latest first try that: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
- If that doesn't fix the issue or if you are already on the latest please provide the herdsman debug logging starting from the point where you start z2m till it fails.
See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Jens-Wymeersch, 2021-09-21:
Thanks @Koenkk - I'll try it this weekend. If I do the firmware update, and it doesn't work, is there a way to do a roll back ?
Secondly, will I need to pair all my devices again or not ?

Koenkk, 2021-09-21:
>  is there a way to do a roll back

Yes, just reflash the old firmware

> Secondly, will I need to pair all my devices again or not ?

No (https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices)

Jens-Wymeersch, 2021-09-21:
Thank you @Koenkk. One last question, is there a way to find out the version of the firmware via HA/addon ? Or if not, when I connect it to RPI, what command should I use to find out which is the version of the firmware.

Koenkk, 2021-09-21:
Its logged on startup and can also be found in the frontend:

[image]

Jens-Wymeersch, 2021-09-25:
@Koenkk Koen, thank you for all your advice. I've just flashed the ZZH with the new firmware and now I can't get it to work anymore. This has become a critical system in my house and hence I need urgent help !

Here are the logs : `Zigbee2MQTT:error 2021-09-25 10:07:39: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-09-25 10:07:39: Failed to start zigbee
Zigbee2MQTT:error 2021-09-25 10:07:39: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-09-25 10:07:39: Exiting...
Zigbee2MQTT:error 2021-09-25 10:07:39: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.js:66:27)
    at Controller.start (/app/lib/controller.js:110:27)
    at start (/app/index.js:95:5)`

Koenkk, 2021-09-25:
> Initially it didn't want to start up, but finally it came to life. Is this something I need to worry about ?

`SRSP - SYS - ping` means the adapter cannot be reached, if it starts fine now and you can restart z2m without problem I wouldn't worry

Jens-Wymeersch, 2021-09-25:
@Koenkk thank you so much. Finally, after a day getting all the routers back into Z2M, I'm all set !

closed 2021-09-26 by Jens-Wymeersch