Koenkk/zigbee2mqtt #8816: Silvercrest Garden Plug is not correct recognized

BigDi, 2021-09-20:
Bought this device
[Silvercrest Garden Plug](https://www.lidl.de/p/silvercrest-gartenspiess-mit-2-steckdosen-zigbee-smart-home/p100329849)
and it is recognized as TuYa TS0101

I work with iobroker zigbee and the issue is there marked as zigbee herdsman converters problem.
It seems that the garden plug is a variation of a TuYa TS 0101.

McMornan, 2021-09-24:
Hi, I can confirm that issue. However I think this is due to it not being in the database :) At least I did not find it in supported devices. Same issue for this outdoor plug: https://www.lidl.de/p/silvercrest-aussen-steckdosen-adapter-zigbee-smart-home/p100333688

BigDi, 2021-11-22:
[image]

BigDi, 2021-11-22:
It seems to be a Tuya device. How to go on?

BigDi, 2021-12-01:
With changes in the devices/lidl.js I can control the plug. Only the icon is missed
{
     fingerprint: [{modelID: 'TS0101', manufacturerName: '_TZ3000_br3laukf'},
         {modelID: 'TS0101', manufacturerName: '_TZ3000_br3laukf'}],
     model: 'TS0101_HG06620',
     vendor: 'Lidl',
     description: 'Garden Spike with 2 sockets Smart Home',
     extend: extend.switch(),
     whiteLabel: [{vendor: 'Lidl', model: 'HG06620'}],
 },

Tonvandermark, 2021-12-10:
Same problem here:
[image]
Is there any planning to add this device to the supported database?

closed 2022-01-17 by github-actions[bot]