Koenkk/zigbee2mqtt #8828: linkquality not saved to state.json

anyuta1166, 2021-09-21:
## What happened

After updating to the latest version of zigbee2mqtt I've noticed that linkquality is no longer saved to state.json.

I don't know is this an intentional change or not, but at least there is some inconsistency. Existing values are read from the file on zigbee2mqtt restart but they are never updated.

## What did you expect to happen

If this is a bug then linkquality should be saved to state.json.

If this is an intentional change then existing values (saved by an older zigbee2mqtt version) should be removed from state.json and should not be read from the file on restart.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC2652P
Adapter firmware version:

sjorge, 2021-09-21:
AFAIK linkquality and last_seen are never saved in state.json, don't think they ever were.

anyuta1166, 2021-09-21:
> AFAIK linkquality and last_seen are never saved in state.json, don't think they ever were.

linkquality was saved in previous versions. It is still present in state.json for devices that were added before zigbee2mqtt update, but the value is no longer updated.

```
    "0x00158d0003e6fa9b": {
        "battery": 100,
        "linkquality": 55,
        "voltage": 3032
    },
```

sjorge, 2021-09-21:
I want back through my backups to check and could go back as far as 1.15.0 and it was is none of them, so if it was before that if it was included at some point

McGiverGim, 2021-09-22:
I can confirm.  My first 17 devices of a total of 26 have a linkquality in the `states.json` file. I suppose it was removed at some point but older devices have the linkquality here.

Masterz69, 2021-09-23:
Have no linkqualty in ```states.json``` nor for devices added last week (Z2M 1.21), nor 6 months ago (Z2M 1.18).

closed 2021-11-01 by github-actions[bot]