Koenkk/zigbee2mqtt #8830: Lights suddenly cannot be controlled

hmorkemo, 2021-09-21:
## What happened
I have many Philips and Ikea lights.
Sometimes they seem to work fine for a while after being added. But suddenly I get this error message and it's impossible to control the light.
Zigbee2MQTT:error 2021-09-21 11:39:37: Publish 'set' 'state' to 'Soveromsterrasse' failed: 'Error: Command 0x0017880103c0399a/11 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'.
To make them available again, I have to delete it from zigbee2mqtt and add it again.

## What did you expect to happen
I expect to be able to control the lights.
## How to reproduce it (minimal and precise)
I just add a light and wait a while. Then it disappears.
It's notable that there are just a few of my lights that behave like this (Both Ikea and Philips). Most of them are very stable.
## Debug info
Zigbee2MQTT version:
Zigbee2mqtt 1.21.1-1, HA Core 2021.9.6, ConBee II. Channel 25. 49 controllers, 50 other devices.
My WiFi routers are on channels 1, 4 and 7.

Adapter firmware version: 0x26720700

Tonio16, 2021-09-22:
Hello

At least, there is a problem for ikea devices. There are lot of posts about set or get state issues recently.

Cheers

Antoine

McMornan, 2021-09-24:
Hi, I have these issues as well. Ususally a power on/off seems to help. To prevent issues with WAF I bind those lamps to a switch nearby.

Tonio16, 2021-09-24:
For a Hue bulb I have also this strange state ""
`debug Received MQTT message on 'zigbee2mqtt/Ampoule ch Nicolas/get' with data '{"state":""}'
debug Publishing get 'get' 'state' to 'Ampoule ch Nicolas'`

jeppper, 2021-10-21:
please let me know if it helps

tanordheim, 2021-11-08:
Sorry for the late response, but I reset both my Hue-bars a couple of weeks ago and re-paired them to zigbee2mqtt. I haven't used them for about a week, but when I sat down at my desk this morning and tried turning them on - one of them works, and one of them doesn't. The faulty one fails with the same error as mentioned before in this thread, `Error: Command ..... failed (no response received)`.

jeppper, 2021-11-08:
None of my bulbs wants to update via OTA.
I also power cycle bulbs not responding but it sometimes helps for a little time but returns unreachable again.

Strange. I wonder if a faulty/instable coordinator would cause this behaviour

closed 2021-12-17 by github-actions[bot]