Koenkk/zigbee2mqtt #8834: QS-Zigbee-D02-TRIAC-LN zigbee dimmer is in the known devices, but is discovered as an unsupported model

adekloet, 2021-09-21:
Hi, 

Device:    modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' is found as unsupported, but seems completly identical to: 

https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-LN.html

Also has the same model name printed on the device.

Can it be made that it's auto-discovered as the Lonsonho QS-Zigbee-D02-TRIAC-LN

--> debug on/off
Received message from unsupported device with Zigbee model 'TS110E' and manufacturer name '_TZ3210_ngqk6jia'

debug Received Zigbee message from '0xa4c1383cc3243cb8', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug No converter available for 'QS-Zigbee-D02-TRIAC-LN' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":0}'
debug Received Zigbee message from '0xa4c1383cc3243cb8', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0

--> debug Brightness control data:

debug Received Zigbee message from '0xa4c1383cc3243cb8', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":12}' from endpoint 1 with groupID 0

db info:

{"id":30,"type":"Router","ieeeAddr":"0xa4c1383cc3243cb8","nwkAddr":38916,"manufId":4417,"manufName":"_TZ3210_ngqk6jia","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[5,4,6,8,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110E","manufacturerName":"_TZ3210_ngqk6jia","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1632236756746}

## Information about the device + link
..

## data/database.db entry of the device
..

adekloet, 2021-09-22:
Thanks,

I will load the Dev Branch asap and report back.

adekloet, 2021-09-22:
Hi The device is now recognised in the Dev Branch, on/off works perfectly. Dimming via the GUI in Z2MQTT does not work unfortunatly, it works through the device and I captured the payload data from the device to zigbee, maybee You could take a look if that could be fixed?

 Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":141}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":63}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":12}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":12}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":75}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":137}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":200}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":263}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":325}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":388}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":451}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":514}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":576}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":639}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":702}' from endpoint 1 with groupID 0
debug Saving state to file /config/zigbee2mqtt/state.json
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":765}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":827}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":890}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":953}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":996}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":996}' from endpoint 1 with groupID 0

that was from off to maximum brightness via the device

adekloet, 2021-09-22:
To add, at boot of the device it gives the following data to zigbee:
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genOnOff', data '{"moesStartUpOnOff":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"64515":10}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"64516":1000}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0xa4c1380acc588695', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":996}' from endpoint 1 with groupID 0

Current on/off
'genLevelCtrl', data '{"64515":10}' from endpoint 1 with groupID   (could this be min value brightness)
'genLevelCtrl', data '{"64516":1000}                                               (could this be max value brightness)
'genLevelCtrl', data '{"61440":996}                                                 (Could this be current brightness set)

Koenkk, 2021-09-22:
In that case the it seems a non-standard zigbee command is required to control the brightness. It will require sniffing with the original gateway to see what to send. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

I will temporarily revert the support.

adekloet, 2021-09-23:
"result": {
"category": "tdq",
"functions": [
{
"code": "switch_led_1",
"type": "Boolean",
"values": "{}"
},
{
"code": "sleep_mode",
"type": "Raw",
"values": "{}"
},
{
"code": "wakeup_mode",
"type": "Raw",
"values": "{}"
},
{
"code": "countdown_1",
"type": "Integer",
"values": "{\"unit\":\"s\",\"min\":0,\"max\":86400,\"scale\":0,\"step\":1}"
},
{
"code": "work_mode",
"type": "Enum",
"values": "{\"range\":[\"light_white\"]}"
},
{
"code": "led_type_1",
"type": "Enum",
"values": "{\"range\":[\"led\",\"incandescent\",\"halogen\"]}"
},
{
"code": "bright_value_1",
"type": "Integer",
"values": "{\"min\":10,\"max\":1000,\"scale\":0,\"step\":1}"
},
{
"code": "brightness_min_1",
"type": "Integer",
"values": "{\"min\":10,\"max\":1000,\"scale\":0,\"step\":1}"
},
{
"code": "brightness_max_1",
"type": "Integer",
"values": "{\"min\":10,\"max\":1000,\"scale\":0,\"step\":1}"
},
{
"code": "relay_status",
"type": "Enum",
"values": "{\"range\":[\"off\",\"on\",\"memory\"]}"
}
],
"status": [
{
"code": "sleep_mode",
"type": "Raw",
"values": "{}"
},
{
"code": "wakeup_mode",
"type": "Raw",
"values": "{}"
},
{
"code": "switch_led_1",
"type": "Boolean",
"values": "{}"
},
{
"code": "work_mode",
"type": "Enum",
"values": "{\"range\":[\"light_white\"]}"
},
{
"code": "led_type_1",
"type": "Enum",
"values": "{\"range\":[\"led\",\"incandescent\",\"halogen\"]}"
},
{
"code": "brightness_max_1",
"type": "Integer",
"values": "{\"min\":10,\"max\":1000,\"scale\":0,\"step\":1}"
},
{
"code": "brightness_min_1",
"type": "Integer",
"values": "{\"min\":10,\"max\":1000,\"scale\":0,\"step\":1}"
},
{
"code": "bright_value_1",
"type": "Integer",
"values": "{\"min\":10,\"max\":1000,\"scale\":0,\"step\":1}"
},
{
"code": "relay_status",
"type": "Enum",
"values": "{\"range\":[\"off\",\"on\",\"memory\"]}"
},
{
"code": "countdown_1",
"type": "Integer",
"values": "{\"unit\":\"s\",\"min\":0,\"max\":86400,\"scale\":0,\"step\":1}"
}

adekloet, 2021-09-23:
**Standard instructions sent to the device**

Code 	Name 	Data type 	Value

relay_status 	Power status setting 	Enum 	{“range”:[“off”,“on”,“memory”]}
work_mode 	Mode 	Enum 	{“range”:[“light_white”]}
switch_led_1 	Switch 1 	Boolean 	{}

brightness_min_1 	Minimum brightness 1 	Integer 	{“min”:10,“max”:1000,“scale”:0,“step”:1}

brightness_max_1 	Maximum brightness 1 	Integer 	{“min”:10,“max”:1000,“scale”:0,“step”:1}

bright_value_1 	Brightness value 1 	Integer 	{“min”:10,“max”:1000,“scale”:0,“step”:1}

countdown_1 	Countdown of switch 1 	Integer 	{“unit”:“s”,“min”:0,“max”:86400,“scale”:0,“step”:1}

led_type_1 	Light source type 1 	Enum 	{“range”:[“led”,“incandescent”,“halogen”]}

closed 2021-12-03 by github-actions[bot]