Koenkk/zigbee2mqtt #8836: Frontend map fails with 'Cannot read property' error

Westcott1, 2021-09-21:
## What happened
Frontend map failed to draw.

info  2021-09-21 19:40:29: Starting network scan (includeRoutes 'false')
~ ~ ~ ~
info  2021-09-21 19:41:13: Network scan finished
info  2021-09-21 19:41:13: MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{},"error":"Cannot read property 'ieeeAddr' of undefined","status":"error","transaction":"9rqku-10"}'

I'm guessing that there is some incomplete device data somewhere.
There is nothing obviously wrong (to me!) in the database.
Where should I look?

## Debug info
Zigbee2MQTT version: 1.21.1-dev (this evening)
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210708

Koenkk, 2021-09-21:
Can you update to latest-dev in 1 hour from now and provide the debug logging again?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

tteck, 2021-09-21:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7206051/log.txt)

Line 616

Koenkk, 2021-09-22:
Can you check if it has been fixed?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

tteck, 2021-09-22:
Yes, I'll check. I'm running bare metal, so it should be a little faster

tteck, 2021-09-22:
It is indeed fixed!

closed 2021-09-22 by Koenkk