Koenkk/zigbee2mqtt #8842: Temperature sensor keeps spamming (RH3052)

adriannowaq, 2021-09-22:
I have an issue with RH3052, it's randomly spamming with messages:

```
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2927}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0
Sep 22 09:54:36 raspberrypi npm[6932]: Zigbee2MQTT:debug 2021-09-22 09:54:36: Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3913}' from endpoint 1 with groupID 0

```
My converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};

 const definition = {
    zigbeeModel: ['RH3052'],
    model: 'RH3052',
    vendor: 'TUYATEC-yg5dcbfu',
    description: 'Temperature & humidity sensor with display',
    supports: 'temperature and humidity',
    fromZigbee: [fz.battery, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.humidity(), e.temperature(), e.battery()],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msRelativeHumidity', 'msTemperatureMeasurement', 'genPowerCfg'];
        const payload1 = [{
            attribute: 'measuredValue',
            minimumReportInterval: 0,
            maximumReportInterval: 0,
            reportableChange: 0,
        }];
        const payload2 = [{
            attribute: 'measuredValue',
            minimumReportInterval: 10,
            maximumReportInterval: 10,
            reportableChange: 32767,
        }];
        await bind(endpoint, coordinatorEndpoint, binds);
        await endpoint.configureReporting('msRelativeHumidity', payload1);
        await endpoint.configureReporting('msTemperatureMeasurement', payload2);
    },
};

module.exports = definition;
```

How can I fix it?

danieledwardgeorgehitchcock, 2021-09-22:
try changing the values for payload1... At that setting, any change in the sensor will be reported.. you probably want to chrank up the maximum report interval on payload2 and change your reportable change to something more meaningful...

adriannowaq, 2021-09-22:
Testing right now, and there is no difference 

> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2817}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0
> Debug Received Zigbee message from '0x60a423fffe44d3f3', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6469}' from endpoint 1 with groupID 0

The same situation occurs when I click button on device once, or while paring. It keeps spamming, my CC2531 crashes because of that

danieledwardgeorgehitchcock, 2021-09-22:
Have you removed your device from the network and restarted Zigbee2MQTT then re-added it? It will have to go through the configuration process in order to change the values from what you have set

adriannowaq, 2021-09-22:
Of course I did

danieledwardgeorgehitchcock, 2021-09-22:
Do you have access to the Zigbee2MQTT frontend? could click on the device in question, go to the reporting tab, and post up a screenshot?

adriannowaq, 2021-09-22:
[image]

closed 2021-10-30 by github-actions[bot]