Koenkk/zigbee2mqtt #8848: Tuya door sensor ZD06 (TS0203) battery 'null'

lassieee, 2021-09-22:
Hello,

In my efforts to replace my Xiaomi door sensors, I ordered a Tuya door sensor.
Pairing went fine but the battery attribute shows as 'null'.

The sensor has 'model ZD06' on the back, and the sensor physically looks like the one on this page: https://forum.phoscon.de/t/tz3000-402jjyro-tuya-door-windows-sensor-ts0203-zd06/486

I'm running koenkk/zigbee2mqtt:1.21.1 on docker, with a CC2652R with firmware Z-Stack 3.x.0 20210708

z2m debug logs for the pairing process:
https://pastebin.com/cGMABTCU

The debug logs are actually from a dev setup with a CC2531 and the sensor as the only connected device.

Some screenshots:

[image]
[image]
[image]

Even though the sensor has a 'battery_low' attribute, it would be nice if it would also give a battery percentage.

Is there something I can do to test? Perhaps use a different converter?

Kind regards,
Casper

deosrc, 2021-09-23:
I've had this on Xiaomi door sensors and also smartthings smart plugs. For me it only occurred when the device was first paired. Turning the device off and on again fixed it. For battery devices, this can usually be done by short pressing the button if there is one.

lassieee, 2021-09-23:
Thanks, I've tried that, but unfortunately the battery remains 'null'. After the tests I did yesterday, I removed the device from my network. I'll leave it connected now, perhaps it needs some time to settle?

three button clicks and one open/close action:
info MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffe0e4d7a', payload '{"battery":null,"battery_low":false,"contact":false,"last_seen":"2021-09-23T20:32:17+02:00","linkquality":30,"tamper":false}'
info MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffe0e4d7a', payload '{"battery":null,"battery_low":false,"contact":false,"last_seen":"2021-09-23T20:32:19+02:00","linkquality":51,"tamper":false}'
info MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffe0e4d7a', payload '{"battery":null,"battery_low":false,"contact":false,"last_seen":"2021-09-23T20:32:21+02:00","linkquality":45,"tamper":false}'
info MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffe0e4d7a', payload '{"battery":null,"battery_low":false,"contact":true,"last_seen":"2021-09-23T20:32:22+02:00","linkquality":51,"tamper":false}'

kevincaradant, 2021-09-26:
Hi
It's not the same device but I think we have the same issue with Tuya devices and the "null" battery level.
https://github.com/Koenkk/zigbee2mqtt/issues/8461

lassieee, 2021-10-01:
Latest version (1.21.2) properly shows battery percentage. Thanks!

closed 2021-10-01 by lassieee