Koenkk/zigbee2mqtt #8849: scene_add with <1 transition does not work.

Slurpgeit, 2021-09-22:
## What happened
I've been trying to add a scene to a group with a transition of 0.5 seconds. However, any transition below 1 second will cause the lights to ignore the transition. Above 1 second everything works. 10.5 seconds for example works just fine.

A 0.5 second transition works just fine when not using scenes.

## What did you expect to happen
The scene turns on with a 0.5 second transition.

## How to reproduce it (minimal and precise)
Add a scene with a 0.5 second transition time.

## Debug info
Zigbee2MQTT version: 1.21.1-1 (HA Supervisor Add-On)
Adapter hardware: CC2652P
Adapter firmware version: CC1352P2_CC2652P_other_coordinator_20210708.zip

Please let me know what kind of other information you might need.

sjorge, 2021-09-23:
From the ZCL
[image]

```
The amount of time, in seconds, it will take for the device to change from its current state to the requested scene.
```

Looks like the smallest value you can enter is 1 second

sjorge, 2021-09-23:
Doesn't look like the field is available in herdsman
https://github.com/Koenkk/zigbee-herdsman/blob/6bf7e974a0fe9a663877217a590184e306884139/src/zcl/definition/cluster.ts#L254-L264

sjorge, 2021-09-23:
@Koenkk  so the listing in herdsman for add matches
[image]

Which does not mention this `TransitionTime100ms` from https://github.com/Koenkk/zigbee2mqtt/issues/8849#issuecomment-925761387 so I don't think we can set it lower than 1 sec?

Slurpgeit, 2021-09-23:
Well that at least explains why it doesn't work :). What's weird though is that it "seems" to work for all operations outside of scenes. Or is this just a placebo and is the transition time bumped up to 1 second in reality?

For now I've worked around the issue by setting the transition time as a device specific option for all devices in the scene. It works in the sense that it neatly transitions from off to on, but I'm not fast enough to tell if it's either 0.5 sec or 1 sec :)

Koenkk, 2021-09-23:
@sjorge I think it is something new indeed (which means not all bulbs might support this) but it is worth trying,  adding below https://github.com/Koenkk/zigbee-herdsman/blob/6bf7e974a0fe9a663877217a590184e306884139/src/zcl/definition/cluster.ts#L262 `{name: 'transitionTime100ms', type: DataType.uint8},` should do it.

sjorge, 2021-09-23:
That’s what I tried but that just broke scene_add on Hue with BT and Trådfri, had not tried others.

The ZCL pdf in the repo lists 2 tables and the one for scene add does not list it, it also has a different field order as the one a bit above it.

~ sjorge

Koenkk, 2021-09-24:
Then I think its indeed something new and not supported (at least by many).

closed 2021-11-01 by github-actions[bot]