Koenkk/zigbee2mqtt #8853: Docker install Zigbee2MQTT： Error while starting zigbee-herdsman

985958118, 2021-09-23:
I use docker to install Zigbee2MQTT and my adapter is ConBee II.
my step:
run  
`ls -l /dev/serial/by-id`
out 
`lrwxrwxrwx 1 root root 13 Sep 21 21:27 usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2420540-if00 -> ../../ttyACM0`
run
`docker run -it -v /home/hyt/docker/zigbee-mqtt/data:/app/data --device=/dev/ttyACM0 -e TZ=Asia/Shanghai -v /run/udev:/run/udev:ro --privileged=true koenkk/zigbee2mqtt
`
and it's wrong
```
Using '/app/data' as data directory
Zigbee2MQTT:info  2021-09-23 15:00:43: Logging to console and directory: '/app/data/log/2021-09-23.15-00-42' filename: log.txt
Zigbee2MQTT:info  2021-09-23 15:00:43: Starting Zigbee2MQTT version 1.21.1 (commit #eb5750d)
Zigbee2MQTT:info  2021-09-23 15:00:43: Starting zigbee-herdsman (0.13.138)
CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2021-09-23 15:00:53: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-09-23 15:00:53: Failed to start zigbee
Zigbee2MQTT:error 2021-09-23 15:00:53: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-09-23 15:00:53: Exiting...
Zigbee2MQTT:error 2021-09-23 15:00:53: undefined
```
my configuration.yaml
```
# Home Assistant integration (MQTT discovery)
homeassistant: false
# allow new devices to join
permit_join: true
# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://ip:1883'
  # MQTT server authentication, uncomment if required:
  user: user
  password: password
  client_id: 'zigbee2mqtt_client'
# Serial settings
serial:
  # Location of CC2531 USB sniffer
  adapter: deconz
  port: /dev/ttyACM0
advancd:
  log_level: debug
```
I don't know why ,please help ,thanks

tteck, 2021-09-23:
You will get help much faster in the [zigbee2mqtt discord channel](https://discord.com/channels/556563650429583360/558344133987663886)

standa4, 2021-09-25:
Hi,
I have the same problem (but with different adapter - zzh!). Did you found how to fix it?
As a work around downgrade to version 1.21.0 worked for me. So right now it works but on an older version of docker image.
Thanks.

985958118, 2021-09-26:
> Hi,
> I have the same problem (but with different adapter - zzh!). Did you found how to fix it?
> As a work around downgrade to version 1.21.0 worked for me. So right now it works but on an older version of docker image.
> Thanks.

sorry，I didn‘t fix it，but I change my adapter from  ConBee II to CC2652 ，and the same steps，it works

closed 2021-11-03 by github-actions[bot]