Koenkk/zigbee2mqtt #8854: bridge/devices not being correctly updated after 6 devices are registered

tmolinos, 2021-09-23:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/8846

<div type='discussions-op-text'>

<sup>Originally posted by **tmolinos** September 22, 2021</sup>
Hi!

I'm having a strange issue. I've been using zigbee2mqtt with success, using eltrolama Zigbee stick.
I has now experiencing with the integration plugin with homebridge. The plugin uses the information provided by the zigbee2mqtt/bridge/devices topic.
All went perfect just until I tried to add a 7th device. All first 6 devices appeared on the Home App in my iPhone. The 7th device never showed up, so I tried removing a device, going to a total of 5 devices, and the I added the same device I was trying before (the 7th). And it showed up! So I tried again to add the one I removed, and the same result, it wouldn't show up in the list.
I investigated a little further and using mqttexplorer I monitored the messages to the devices topic. And it seams that zigbee2mqtt stops updating the topping whenever there are more than 6 devices registered.
Scenario 1 - 1 to 6 devices registered
Whenever I change the name of a device, the devices topic get updated
Scenario 2 - 7 or more devices registered
The topic was not updated when I added the device. A change in a random registered device name, does not trigger a device topic update

Any help would be appreciated.

I'm using the latest docker image:

Zigbee2MQTT version
1.21.1 commit: eb5750d
Coordinator type
zStack3x0
Coordinator revision
20210120
Frontend version
0.6.0
</div>

tmolinos, 2021-09-30:
Well it seems I'm the only one having this issue. 
What would be your approach to fix this? Start all over again? I'm using docker by the way.

Koenkk, 2021-09-30:
https://github.com/Koenkk/zigbee2mqtt/pull/8859 might fix this, I will merge it this weekend so please check if you still have this problem with the latest -dev branch after this mr is merged.

tmolinos, 2021-10-04:
Hi Koenkk thanks for your work and answer!
I've just updated my docker container to latest-dev tag.
Unfortunately I'm still experiencing the same issue. Every time I add a seventh device the MQTT topic bridge/devices stops updating. I can reproduce the issue every time.
Last week I tried starting from zero, so I deleted all my setting and setup a new fresh instance of zigbee2mqtt. I had to pair my devices all over again. Still the issue resurfaced after I added the seventh device.

Koenkk, 2021-10-04:
Could you provide the debug log of this?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

tmolinos, 2021-10-05:
I've enabled logging as instructed. Here is the log...

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7285321/log.txt)

I started the container adding the run command parameter for logging
Server started with 6 devices paired
Renamed device from CLOSET_DOOR_2 to CLOSET_DOOR_20 and saw it being reflected on MQTT /bridge/devices topic
Allowed join and added another device 0x00158d00010ec723
Renamed it to DESK_LIGHT, and no update on MQTT
Renamed device from CLOSET_DOOR_2 to CLOSET_DOOR_20 , and no update on MQTT
Renamed device from CLOSET_DOOR_20 to CLOSET_DOOR_2 , and no update on MQTT
Disabled joining and removed DESK_LIGHT
Renamed device from CLOSET_DOOR_2 to CLOSET_DOOR_20 and saw it being reflected on MQTT /bridge/devices topic
Renamed device from CLOSET_DOOR_20 to CLOSET_DOOR_2 and saw it being reflected on MQTT /bridge/devices topic

I hope this can help in some way!

Koenkk, 2021-10-05:
I cannot reproduce the problem yet.

- What MQTT broker are you using? (could it be that your broker has a max message size limit)
- Can you do:
```
docker exec -it ZIGBEE2MQTT_CONTAINER_NAME sh
vi dist/mqtt.js
```

  - Now add the following on line 142 (above `if (!skipLog) {`): `console.log(topic)`
  - Save the file
  - Exit container (execute `exit`)
  - Restart zigbee2mqtt container, (`docker restart ZIGBEE2MQTT_CONTAINER_NAME`) 
  - Reproduce problem again and provide log

tmolinos, 2021-10-05:
Oh men!! It seems to be a configured restriction on the MQTT broker!! It had a "message_size_limit" parameter! I set it to 0 with set it to the protocol maximum, and restarted the broker. After that it all seems to be working now! I'll test a little further tomorrow and I'll let you know!
Thanks!