Koenkk/zigbee2mqtt #8855: Problem with Tuya TS004F

skapced, 2021-09-23:
Hello,
I have no feedback from this 4 button switch while it is well supported by Zigbee2mqtt.
(https://www.zigbee2mqtt.io/devices/TS004F.html)
when I press the button, I have no information in the log with this equipment

[image]

zigbee2mqtt/database:
`{"id":8,"type":"EndDevice","ieeeAddr":"0x84fd27fffe8d4144","nwkAddr":17356,"manufId":4098,"manufName":"_TZ3000_xabckq1v","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_xabckq1v","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00158d0000ce21c3","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00158d0000ce21c3","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00158d0000ce21c3","endpointID":1},{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00158d0000ce21c3","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1217940983},"lastSeen":1632405051186}`

## Debug info
Zigbee2MQTT version: 1.21 on JEEDOM
Adapter hardware: Zigate USB
Adapter firmware version:

Squixx, 2021-09-30:
I have exactly the same problem.

Device interviews fine. But no response whatsover on any button

MattWestb, 2021-10-01:
I think (= not knowing) that Z2M have not setting up binding and  reporting for all clusters so the device is not reporting the commands sent to the coordinator.

skapced, 2021-10-01:
do you know the procedure to add them manually?

MattWestb, 2021-10-01:
No in Z2M in ZHA the system is making the binding and attribute reporting OK.

skapced, 2021-10-01:
ok, would it be possible to go into Z2M (on this type of equipment) and put several screenshots please (link, reports and parameters)? I would like to try to include them manually

Squixx, 2021-10-01:
Looking further into this

It seems my modelnumber is
```
{vendor: 'Haozee', model: 'ESW-OZAA-EU'}],
```

However, even when setting an external converter that's equal to

```
        zigbeeModel: ['TS0044'],
        model: 'TS0044',
        vendor: 'TuYa',
        description: 'Wireless switch with 4 buttons',
        whiteLabel: [{vendor: 'Lonsonho', model: 'TS0044'}, {vendor: 'Haozee', model: 'ESW-OZAA-EU'}],
```

I get zero mqtt messages.

So it seems _both_ the device is recognized as "TS004F" (no clue if it anounces wrong, or its recognised wrong) 

while it should be a whitlabel of TS0044.

_and_ in both cases  messages are not correctly recieved.

At this point im quite lost on where to continue with debugging unfortunately :(

bidulle95, 2021-11-05:
NO NEWS ?

MariusLaurinaitis, 2021-11-16:
I also have a problem. Zigbee2Mqtt shows it like Tuya TS004F. The switch behaves strangely: pressing the upper left turns on the other zigbee switches (which are not associated with it in any way), the lower left turns off the enabled switches (which are not associated with it as well). The top right and bottom do nothing. I tried to use the written blueprint automation, it works like I configure, but without the set function the upper left still turns on the other zigbee switches, and the lower left without the assigned function still turns off the turned on zigbee switches.

ravindrajob, 2021-12-14:
Hello, any update about this issue ? :)

skapced, 2021-12-15:
> Hello, any update about this issue ? :)

hello, i tested only once but apparently, the last update rectify this problem.
I must continue the tests to validate the operation

my configuration:
Zigbee2MQTT version: 1.22.1 on Home Assistant (HAOS)
Adapter hardware: Zigate USB
Adapter firmware version: 1E03

closed 2022-01-22 by github-actions[bot]