Koenkk/zigbee2mqtt #8858: OTA stays at 0,00%

Tonio16, 2021-09-23:
## What happened
Either for philipps or ikea devices, the OTAs are stuck to 0.00%.

## What did you expect to happen
OTA to be completed.

## How to reproduce it (minimal and precise)
I tried on two different Pi, a 2 and 3+, and different version of z2m. My zigate is typically  plug to an usb hub with an usb cable but I tried other configuration without succes

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: zigate v1
Adapter firmware version: 1d03

Log: 
[log_debug-ota.log](https://github.com/Koenkk/zigbee2mqtt/files/7220141/log_debug-ota.log)

Tx

Antoine

McMornan, 2021-09-24:
Hi, I had the same issue. I simply send the OTA command using mqtt cmd line client. After that OTA started normally. Somehow the first request seems to have been lost ...

Tonio16, 2021-09-24:
> Hi, I had the same issue. I simply send the OTA command using mqtt cmd line client. After that OTA started normally. Somehow the first request seems to have been lost ...

Thanks but I have no idea on how to do it. I am only using the frontend and I didn’t find how to send such cmd line from frontend.

Antoine

McMornan, 2021-09-24:
Hi,

open a pi cmdline bash and enter  
mosquitto_pub -h IPOFYOURMQTTSERVER -p 1883 -t zigbee2mqtt/bridge/request/device/ota_update/update -m {"id":"ID OF YOUR DEVICE"}'

if not already done enter "sudo apt install mosquitto-clients"

Tonio16, 2021-09-24:
I enter the line you gave me with the good IP and ID but I have nothing more than a prompt: >
I will have a look to mqtt explorer, to see if I can do better with it.

Antoine

McMornan, 2021-09-24:
Yes of course you do not see the result there. You need to open a second bash window with this line:
mosquitto_sub -p 1883 -h IPOFYOURMQTTSERVER -t '#' -v | grep zigbee2mqtt

There you will see the response of zigbee2mqtt

Tonio16, 2021-09-25:
Hello 

If I enter:
`mosquitto_pub -h 192.168.1.xxx -p 1883 -t zigbee2mqtt/bridge/request/device/ota_update/update -m {"id":"0x0017880108217905"}'
`
I have:
`info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"{id:0x0017880108217905}\nmosquitto_pub -h 192.168.1.130 -p 1883 -t zigbee2mqtt/bridge/request/device/ota_update/update -m {\"id\":\"0x0017880108217905\"}"},"error":"Device '{id:0x0017880108217905}\nmosquitto_pub -h 192.168.1.130 -p 1883 -t zigbee2mqtt/bridge/request/device/ota_update/update -m {\"id\":\"0x0017880108217905\"}' does not exist","status":"error"}'
error Device '{id:0x0017880108217905} mosquitto_pub -h 192.168.1.130 -p 1883 -t zigbee2mqtt/bridge/request/device/ota_update/update -m {"id":"0x0017880108217905"}' does not exist`

Antoine

McMornan, 2021-09-26:
Hi, did you forget the first ' ? 

mosquitto_pub -h 192.168.1.xxx -p 1883 -t zigbee2mqtt/bridge/request/device/ota_update/update -m '{"id":"0x0017880108217905"}'

Tonio16, 2021-09-26:
> 
> 
> Hi, did you forget the first ' ?
> 
> mosquitto_pub -h 192.168.1.xxx -p 1883 -t zigbee2mqtt/bridge/request/device/ota_update/update -m '{"id":"0x0017880108217905"}'

Ok, it wasn't in your first message and I didn't realized it was needed this '.
I tried but I still have an error at the end.
`info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Liseuse 1er"},"error":"Update of 'Liseuse 1er' failed (Timeout: device did not request any image blocks)","status":"error"}'`

Your trick doesn't help me.

Antoine

McMornan, 2021-09-26:
Hi, 

There are only three more things I can think of:

1.) my french is not that strong, is it a battery powered device ? Then you need to wake it up before sending the ota message (e.g. clicking a battery powered switch continuously to prevent it from falling asleep)
2.) does the device appear on your network map as being connected to the network ? For me sometimes hue and tradfri bulbs drop out and a simple powercycle heals that.
3.) I did some googling on the error message and found this https://github.com/Koenkk/zigbee2mqtt/issues/3098 However that means your stick is not powerful enough ...

closed 2021-11-03 by github-actions[bot]

Tonio16, 2021-12-23:
I change for a controller, a sonoff 3.0.0 itead. No more trouble to update.

Antoine

christian-nils, 2022-01-18:
Hi,

I do experience the same problem. I have a Floalt and a Trådfri remote, none of them is updatable. I tried to send the right command via MQTT, it seems to trigger it blocks at 0.00% with a timeout error after:

```
zigbee2mqtt    | Zigbee2MQTT:info  2022-01-18 12:02:10: Updating '0x90fd9ffffeee701d' to latest firmware
zigbee2mqtt    | Zigbee2MQTT:info  2022-01-18 12:02:19: Update of '0x90fd9ffffeee701d' at 0.00%
zigbee2mqtt    | Zigbee2MQTT:info  2022-01-18 12:02:19: MQTT publish: topic 'zigbee2mqtt/0x90fd9ffffeee701d', payload '{"color_mode":"xy","linkquality":92,"power_on_behavior":null,"state":null,"update":{"progress":0,"state":"updating"},"update_available":null}'
zigbee2mqtt    | Zigbee2MQTT:info  2022-01-18 12:04:49: MQTT publish: topic 'zigbee2mqtt/0x90fd9ffffeee701d', payload '{"color_mode":"xy","linkquality":91,"power_on_behavior":null,"state":null,"update":{"state":"available"},"update_available":null}'
zigbee2mqtt    | Zigbee2MQTT:info  2022-01-18 12:04:49: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x90fd9ffffeee701d"},"error":"Update of '0x90fd9ffffeee701d' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"ewsxp-2"}'
zigbee2mqtt    | Zigbee2MQTT:error 2022-01-18 12:04:49: Update of '0x90fd9ffffeee701d' failed (Timeout: device did not request any image blocks)
```

I have a PiZigate+ on a RPi 4. 

Greetings

christian-nils, 2022-01-18:
Hi, 

It seems that I managed to get the remote updating. I am not sure why. I have click multiple times on the power button and held the power button > 3s . I am not sure what made it work.

Next is to try with the Floalt that failed as well before.

FliesWithWind, 2023-07-18:
I have exact same problem. I'm trying to update Philips HUE Smart Plug 
https://www.zigbee2mqtt.io/devices/929003050601.html#philips-929003050601

I'm using ZigStar LAN Gateway.

`
info  2023-07-18 16:22:42: Updating 'Mk4 Power Plug' to latest firmware
info  2023-07-18 16:22:43: Update of 'Mk4 Power Plug' at 0.00%
info  2023-07-18 16:22:43: MQTT publish: topic 'zigbee2mqtt/Mk4 Power Plug', payload '{"linkquality":112,"state":"ON","update":{"installed_version":16778760,"latest_version":16780034,"progress":0,"state":"updating"},"update_available":null}'
info  2023-07-18 16:25:13: MQTT publish: topic 'zigbee2mqtt/Mk4 Power Plug', payload '{"linkquality":112,"state":"ON","update":{"installed_version":16778760,"latest_version":16780034,"state":"available"},"update_available":null}'
info  2023-07-18 16:25:13: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Mk4 Power Plug"},"error":"Update of 'Mk4 Power Plug' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"fpbhp-3"}'
error 2023-07-18 16:25:13: Update of 'Mk4 Power Plug' failed (Timeout: device did not request any image blocks)
`

fapgomes, 2023-09-06:
I have the same problem, using the ZigStar LAN Gateway.

jose-jacinto, 2023-10-02:
Exactly the same problem, with Zigstar lan gateway updating an ubisys device

mgisbers, 2023-12-29:
My ubisys devices show all the same error. Update process starts and stops with 'Timeout: device did not request any image blocks'.

jjanderson, 2024-01-19:
I was having the same issue with a hue dimmer. Zigbee Model RWL021

What solved it for me was resetting the device (holding down all 4 buttons)

Hope this helps anyone else struggling