Koenkk/zigbee2mqtt #8860: Hue Resonate 1746430P7

GerhardLang, 2021-09-23:
Hello.

I try to get rid of my hue bridge and tried to integrate:
Philips Hue Resonate outdoor wall lamp 1746430P7.

I do have homebridge with homebridge Zigbee2MQTT plugin and zigbee2MQTT.
Hardware is conbee2.

On zigbee2mqtt.io I found the lamp as supportet.

But it appears in device list of Zigbee2MQTT as unsupportet:

Friendly name
0x00178801091df17a
Last seen
N/A
Device type
Router
Zigbee Model
1746447P7
Zigbee Manufacturer
Philips
Support status
Not supported
IEEE Address
0x00178801091df17a
Network address
0x0F1A
Firmware build date
20210331
Firmware version
1.88.1
Power
Interview completed
True

Have I  to go through the process of adding a new device or can I copy / paste something ?

Thank you for a short note so I can help myself better. 

Gerhard

danieledwardgeorgehitchcock, 2021-09-24:
Support for this device has been added however, it may not yet be in the production codebase.

Please switch to Dev / Edge and re-bind your device

GerhardLang, 2021-09-24:
Thank you for the hint. I removed the old zigbe2mqtt installation after saving my data folder.
Then switched to the latest-dev. 
Unfortunately the same happened, the Philips Hue is not recognized as supported device.
Bute all of my greenpower switches in the house were recognized as another manufacturer (eco...) and now all the existing programming is lost, even after restoring my data folder. Grrrmpfff. I hope when the next release is published this will not happen.

Regarding the Philips hue I will wait until next official release and hope that the hue resonate wall lamp is included. Until then I will continue with hue bridge for the three lamps.

Thank you, I tried my best.

Gerhard

closed 2021-11-01 by github-actions[bot]