Koenkk/zigbee2mqtt #8863: Hue White and Colour Ambiance Amarant linear outdoor light

OsteHovel, 2021-09-24:
## Information about the device + link

[image]

https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-amarant-linear-outdoor-light/1746630P7

Working great with provided configuration

## External converter
```Javascript
const { toZigbeeConverters: tz, fromZigbeeConverters: fz, exposes } = require('zigbee-herdsman-converters');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

// Copied from zigbee-herdsman-converters/devices/philips.js with no modifications
const hueExtend = {
  light_onoff_brightness: (options = {}) => ({
    ...extend.light_onoff_brightness(options),
    toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp(options),
    toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_color: (options = {}) => ({
    ...extend.light_onoff_brightness_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_color({ supportsHS: true, ...options }).toZigbee
      .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
  light_onoff_brightness_colortemp_color: (options = {}) => ({
    ...extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options }),
    toZigbee: extend.light_onoff_brightness_colortemp_color({ supportsHS: true, ...options })
      .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
  }),
};

module.exports = [{
  zigbeeModel: ['1746630P7'],
  model: '1746630P7',
  vendor: 'Philips',
  description: 'Hue White and Colour Ambiance Amarant linear outdoor light',
  meta: { turnsOffAtBrightness1: true },
  extend: hueExtend.light_onoff_brightness_colortemp_color(),
  ota: ota.zigbeeOTA,
}];
```
The hueExtend const is directly copied from Philips.js with no modifications so its only the bottom part that needs to be added to Philips.js

## data/database.db entry of the device

```JSON
{
  "id": 2,
  "type": "Router",
  "ieeeAddr": "0x00178801091e019a",
  "nwkAddr": 41666,
  "manufId": 4107,
  "manufName": "Philips",
  "powerSource": "Mains (single phase)",
  "modelId": "1746630P7",
  "epList": [
    11,
    242
  ],
  "endpoints": {
    "11": {
      "profId": 260,
      "epId": 11,
      "devId": 269,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        4096,
        768,
        64513
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "1746630P7",
            "manufacturerName": "Philips",
            "powerSource": 1,
            "zclVersion": 2,
            "appVersion": 2,
            "stackVersion": 1,
            "hwVersion": 0,
            "dateCode": "20190529",
            "swBuildId": "1.56.7_r29434"
          }
        },
        "lightingColorCtrl": {
          "attributes": {
            "colorCapabilities": 31,
            "colorTempPhysicalMin": 153,
            "colorTempPhysicalMax": 500
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 2,
  "stackVersion": 1,
  "hwVersion": 0,
  "dateCode": "20190529",
  "swBuildId": "1.56.7_r29434",
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {
    "configured": 88764544
  },
  "lastSeen": 1632467839342
}
```

OsteHovel, 2021-09-24:
Color temp range
```{"read": {"cluster": "lightingColorCtrl", "attributes": ["colorTempPhysicalMin", "colorTempPhysicalMax"]}}```
Response:
`debug Received Zigbee message from '0x00178801091e019a', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":500,"colorTempPhysicalMin":153}' from endpoint 11 with groupID 0`

danieledwardgeorgehitchcock, 2021-09-24:
Do you want me to add it to the codebase?

This is something you can do yourself if you are reasonably competent with Git?

danieledwardgeorgehitchcock, 2021-09-24:
I have created the a pull request for you.

Once it is merged, please switch to the development / edge version of Zigbee2MQTT in order to use the device without your external converter

closed 2021-09-24 by Koenkk