Koenkk/zigbee2mqtt #8865: Tuya TS011F, manufacturer _TZ3000_dpo1ysak

Z18Aplha, 2021-09-24:
## Information about the device + link

Zigbee2MQTT:warn  2021-09-24 12:05:06: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_dpo1ysak'

One of the Tuya Smart plugs with power measure.
Seems to be another manufacturer id.

Link to Aliexpress: [https://de.aliexpress.com/item/1005002344798281.html?spm=a2g0s.9042311.0.0.1de44c4dqEgoVe](https://de.aliexpress.com/item/1005002344798281.html?spm=a2g0s.9042311.0.0.1de44c4dqEgoVe)

## data/database.db entry of the device
{"id":13,"type":"Router","ieeeAddr":"0x8cf681fffef5058e","nwkAddr":29194,"manufId":4098,"manufName":"_TZ3000_dpo1ysak","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_dpo1ysak","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1632477506134}

JohnRabbit1, 2021-09-24:
Hi
I received this thing as well today. After some experiments i found out it behaves more like a ts0121 device???? So i added a fingerprint in the section for this TS0121 device (fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dpo1ysak'}],) 
And it seems to work (i can switch it on and off) It reports power etc but i have not checked if that is accurate.

Koenkk, 2021-09-24:
Has just been added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3076

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-09-24 by Koenkk

Z18Aplha, 2021-09-25:
> Hi
> I received this thing as well today. After some experiments i found out it behaves more like a ts0121 device???? So i added a fingerprint in the section for this TS0121 device (fingerprint: [...]

based on this post, i assume the plug should be able to measure the power. The current implementation doesn't provide power management. Is it possible to implement the plug like @JohnRabbit1 did?

Z18Aplha, 2021-09-25:
@JohnRabbit1 Do you get current and power values or just voltage?
I did the same (adding fingerprint to TS0121) and received values for power measurement as well.
Referring to to documentation of z2m the device push its power measurements every 60 seconds (no /get possible). 
After turning on (a light net) i waited (much more than 60 seconds) but the values for current and power stays zero.
But i can confirm, the plug sends something every 60 seconds because the LQI updates frequently and the voltage changes (+/-1V).
Sometimes the current is 0.01A, but thats too low for sure. 

Did you get the same behavior of the plug or is it just mine?

fidodido1202, 2021-09-26:
Nice it works with the dev branch :) received the same plug yesterday great work!

cabrust, 2021-09-26:
After the merge of [this PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3082), I get the same behavior as @Z18Aplha, with the same error messages. All measurements are reported as "null". Reading the power manually works fine though:
[image]

Z18Aplha, 2021-09-26:
yeah the device supports power measurement for sure. 
Go to [Koenkk/zigbee-herdsman-converters#3076](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3076). There i reported more details. Using the plug as TS0121 provides the power and current values in the right way.

fidodido1202, 2021-09-28:
Also received this plug, and with the dev branch the smart plug is reconized (thanks for that :) ) and i have now the following components
**urrent, energy, linkquality, power, power_outage_memory, switch, voltage**

In the logs file of the container i see the device as followed, i suppose this is correct?
"manufacturerID":4098,"manufacturerName":"_TZ3000_dpo1ysak","model":"TS011F_plug","modelID":"TS011F","networkAddress":50415,"powerSource":"Mains (single phase)","type":"Router","vendor":"TuYa"}],"type":"devices"}',

But inside home assistant with the power monitoring it still shows **None** as value.
Will this be fixed in a new version of the image? Or do i need to change the fingerprint anyway?
I am quiet new with all of this not sure how to change the fingerprint to get this to work.

oroszgergely, 2021-11-06:
If I press the physical OnOff button on the plug, it does not automatically reported to Z2M, it stays as it was, however i can read out the actual value on the dev console:
[image]

device info:
[image]