Koenkk/zigbee2mqtt #8868: IKEA TRÅDFRI Repeater (E1746) stops responding daily.

Wh1terat, 2021-09-24:
My coordinator is a Xiaomi Gateway 3 running the z2m firmware.
(https://github.com/AlexxIT/XiaomiGateway3#zigbee2mqtt-mode)

I've got 2x E1746 extending my Zigbee network, both running 2.2.005.

After around 24 hours they simply stop responding and require rebooting, leaving the end devices dangling until rebooted (even though they can _just_ reach the coordinator directly)

```
Error Failed to execute LQI for 'Landing Repeater'
Error Failed to execute LQI for 'Garage Repeater'
```

I've seen that there is an OTA update available - v2.3.070. 
This is only listed on the prod manifest (2.2.005 is latest on test) - but 2.3.070 _is_ available on test and the file does differ from that on prod but the device will still not update using it.

```
http://fw.ota.homesmart.ikea.net/global/GW1.0/01.15.055/bin/10037603-3.1-TRADFRI-signal-repeater-2.3.070.ota.ota.signed
http://fw.test.ota.homesmart.ikea.net/global/GW1.0/01.15.055/bin/10037603-3.1-TRADFRI-signal-repeater-2.3.070.ota.ota.signed
```
(202704 bytes vs 202376)

Any ideas?

McMornan, 2021-09-24:
Hi, I have one of those repeaters running 2.3.070 firmware (updated on tradfri gateway) The repeater is running fine with 1.21.1 an reporting back every 4 hours. Maybe you could try to update the firmware using a tradfri gateway ?

Wh1terat, 2021-09-24:
Unfortunately the _only_ TRÅDFRI products I have are the repeaters and OTA updates for 2.3.070 within z2m aren't working :/

Wh1terat, 2021-09-24:
Unfortunately there is no IKEA local to me so with shipping a gateway would cost 4.5x what the repeater did.

Would be nice to understand why the GBL updates don't work from z2m.

Wh1terat, 2021-10-05:
Gave up and recycled a spare CC2531 dongle as a router and removed the E1746s.
Works perfectly, network been up several days without drops.

So the issue is definitely something specific to the E1746, possibly firmware but since OTA doesn't work for the latest version I cannot confirm.

MattWestb, 2021-10-06:
I think better using the CC-2531 and installing ZHA in HA and doing the OTA update for the repeater then the CC-2531 have very bad hardware both CPU and radio.

Wh1terat, 2021-10-06:
Interesting that you say ZHA is capable of OTA on the IKEA products? Is this confirmed?
Don't want to disrupt my now working setup and reflash the CC2531 stick if not 100%.

Also I have some of the Sonoff CC2652P based sticks on back order :)

MattWestb, 2021-10-06:
Its working very well and its possible doing downgrading of light for getting the "IKEA default scenes" in there lights.
https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates
And all current firmware in the production and revue of coming firmware https://github.com/zigpy/zigpy/discussions/660.

Install ZHA in HA and updating your devices and then you can uninstalling it and using Z2M and flashing your CC.2531 with router firmware and using it 2.

closed 2021-11-13 by github-actions[bot]