Koenkk/zigbee2mqtt #8871: E1812 unwanted pairing to 2*LED1836G9

hank-droid, 2021-09-24:
## What happened
when I press either E1812, both LED1836G9 switch on
## What did you expect to happen
nothing
## How to reproduce it (minimal and precise)
pairing sequence:

1. IKEA ICPSHC24-10EU-IL-1, then unplugged off
2. first E1812
3. second E1812
4. first LED1836G9
5. E1524/E1810 
        then paired to first LED1836G9, 
         first LED1836G9 unplugged
6. second LED1836G9
then all plugged back

second try pairing sequence:

1. IKEA ICPSHC24-10EU-IL-1, then unplugged
2. first LED1836G9
3. E1524/E1810 
        then paired to first LED1836G9, 
         first LED1836G9 unplugged
4. second LED1836G9, then unplugged
5. first E1812
6. second E1812
then all plugged back

issue occurs also with zigbee2mqtt add-on stopped
it seems that as soon as E1812 or N2 join the zigbee2mqtt, then their 

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2652P
Adapter firmware version: not sure, was flashed upon purchase with latest Koenkk firmware a week ago

## herdsman debug logging

Zigbee2MQTT:info  2021-09-24 19:45:44: MQTT publish: topic 'zigbee2mqtt/Lava/action', payload 'on'
2021-09-24T17:45:44.895Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-09-24T17:45:44.895Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-09-24T17:45:44.895Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-09-24T17:45:44.896Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-09-24T17:45:44.896Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-24T17:45:45.917Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,3,197]
2021-09-24T17:45:45.917Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,3,197]
2021-09-24T17:45:45.917Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,3] - 197
2021-09-24T17:45:45.918Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3}
2021-09-24T17:45:45.918Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

## Notes
same happened on edge version, in which case also the IKEA N2 remote (STYRFRY, the newest model) after being paired to zigbee2mqtt and then to IKEA ICPSHC24-10EU-IL-1, pressing the on/brightness up button would switch on both the LED1836G9
reverted to 1.18-1 version, reset and removed IKEA N2 remote, paired IKEA N2 remote only with IKEA ICPSHC24-10EU-IL-1,, issue not occurring with N2
issue occurs with zigbee2mqtt add-on stopped as well, as long as the E1812 or N2 were joined to zigbee2mqtt

it seems that as soon as E1812 or N2 join the zigbee2mqtt, then their button (on in case of N2) switches on the LED1836G9
on cc2531, it would also interfere with the CPSHC24-10EU-IL-1

hank-droid, 2021-09-24:
i am not so concerned about the E1812, but that it can occur to the N2 if it would be added to stable version, since i plan to have 10 of them in my set up and i would like them to be in the zigbee2mqtt network, rather than only paired to the bulbs to avoid this conflict.
thank you!

hank-droid, 2021-09-25:
i run multiple tests with the ICPSHC24-10EU-IL-1 and N2 remote as well, in genera it seems that if i join them to the zigbee coordinator, then they all act as if they are part of the same group.

closed 2021-11-03 by github-actions[bot]

BalassaMarton, 2023-01-16:
I have the exact same problem with E1743 (on/off switch), LED1842G3 (bulb) and ICPSHC24-10EU-IL-1 (10w LED driver). I even created bind groups for each so that they are not in the same group, but no success. The LED driver and bulb can be controlled separately from UI and MQTT, but not using the switch. I have multiple switches, some bound to the LED driver, others to the bulb, but they all control both lights, no matter what I do. Removing the switches from the default bind group always fails.

BalassaMarton, 2023-01-16:
Update - I was able to remove SOME of the dimmers from the default bind group, then added them to the respective groups (one for each light), but they still switch both lights.

BalassaMarton, 2023-01-16:
Solved! I did the below steps in this exact order (after force-removing the devices):

1. Add the lights
1. Create a dedicated group for each light
3. Put each light into its dedicated group
5. Add the first switch
6. Bind the switch to its respective group, without removing from the default bind group
7. Go to Dashboard, verify that the switch is controlling the correct light
8. Repeat from step 4 for remaining switches

Note: my switches have version 2.2.010 firmware, which appears as the latest as of today.