Koenkk/zigbee2mqtt #8881: Help with 'TS011F' and manufacturer name '_TZ3000_x2ddiwsv'

mmspock, 2021-09-25:
Hi, 

I purchased following plug: 
https://allegro.pl/oferta/gniazdko-wtyczka-zigbee-3680w-16a-tuya-pomiar-pr-10538098315

It reports as: Device '0xb4e3f9fffe206ffb' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_x2ddiwsv' is NOT supported

Honestly I did not try to go through the guide of adding new device. I have seen other similar threads therefore I assume that this might be just adding new manufacturer name.   I am on version 1.21.1. 

Regards

danieledwardgeorgehitchcock, 2021-09-27:
Hi,

If you could post your database.db entry for this device, we could advise on whether extending an existing device handler, or creating a new one is required.

mmspock, 2021-10-02:
@danieledwardgeorgehitchcock this entry looks like that

`{
   "id":2,
   "type":"Router",
   "ieeeAddr":"0xb4e3f9fffe206ffb",
   "nwkAddr":49847,
   "manufId":4098,
   "manufName":"_TZ3000_x2ddiwsv",
   "powerSource":"Mains (single phase)",
   "modelId":"TS011F",
   "epList":[
      1,
      242
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":266,
         "inClusterList":[
            0,
            3,
            4,
            5,
            6,
            57344,
            57345
         ],
         "outClusterList":[
            25,
            10
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"TS011F",
                  "manufacturerName":"_TZ3000_x2ddiwsv",
                  "powerSource":1,
                  "zclVersion":3,
                  "appVersion":67,
                  "stackVersion":0,
                  "hwVersion":1,
                  "dateCode":""
               }
            }
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[
            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":67,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1632573982786
}`

mmspock, 2021-10-02:
@danieledwardgeorgehitchcock this entry looks like that

{
   "id":2,
   "type":"Router",
   "ieeeAddr":"0xb4e3f9fffe206ffb",
   "nwkAddr":49847,
   "manufId":4098,
   "manufName":"_TZ3000_x2ddiwsv",
   "powerSource":"Mains (single phase)",
   "modelId":"TS011F",
   "epList":[
      1,
      242
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":266,
         "inClusterList":[
            0,
            3,
            4,
            5,
            6,
            57344,
            57345
         ],
         "outClusterList":[
            25,
            10
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"TS011F",
                  "manufacturerName":"_TZ3000_x2ddiwsv",
                  "powerSource":1,
                  "zclVersion":3,
                  "appVersion":67,
                  "stackVersion":0,
                  "hwVersion":1,
                  "dateCode":""
               }
            }
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[
            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":67,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1632573982786
}

closed 2021-11-11 by github-actions[bot]