Koenkk/zigbee2mqtt #8882: configuration.yaml lost entry + json format added

illivanillie, 2021-09-25:
# Bug Report
configuration.yaml lost "last_seen" entry (which I added a few days ago) + 
configuration.yaml had waved brackets like in json-format.
Nothing changed in terms of behavior, only the last_seen attribute was lost.

## What did you expect to happen
The last_seen entry to be permanent

## How to reproduce it (minimal and precise)

See above

## Debug Info
Zigbee2MQTT version: 1.21.1 
Adapter hardware: , CC26X2R1 
Adapter firmware version: zStack3x0

illivanillie, 2021-09-26:
I think it is related to a restart of the host. The yaml file is then replaced by the .bk file and vice versa

illivanillie, 2021-10-15:
Hello, the issue is still present. Could you please help me preventing the change bin configuration.yaml?
It recently happened after a restart of the add on through HA supervisor. The last seen entry is very important for me. I have no idea why it is thrown out..

illivanillie, 2021-10-16:
Just for comparison later, this is my config now. I did NOT add this json-like format. It happened after a restart of zigbee to enable last_seen again.

`{
  "external_converters": [],
  "devices": [
    "devices.yaml"
  ],
  "groups": [
    "groups.yaml"
  ],
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "user": "mqttuser",
    "password": "mqttuserpassword",
    "server": "mqtt://core-mosquitto:1883"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "log_level": "warn",
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "last_seen": "ISO_8601_local",
    "availability_blocklist": [],
    "availability_passlist": [],
    "availability_timeout": 0
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": {
    "port": 8099
  },
  "experimental": {},
  "zigbee_herdsman_debug": true
}
`

Koenkk, 2021-10-16:
This is not the zigbee2mqtt configuration.yaml (this is the configuration from the hass addon)

illivanillie, 2021-10-16:
Oh my goodness me..
On your page it says `data/configuration.yaml`.
Is this the config accessible through the supervisor? 
my lord.. I was in the wrong way I guess

closed 2021-11-25 by github-actions[bot]