Koenkk/zigbee2mqtt #8885: Transmit_power > 5 has no effect on CC2652P

ALaDoffe, 2021-09-26:
## What happened
Hello 

Regarding this issue #8878 I decided to : 

1. Open an issue as suggested on egony repo https://github.com/egony/cc2652p_E72-2G4M20S1E/issues/27

2. migrate my coordinator firmware to CC1352P2_CC2652P_other_coordinator_20210708.zip
        --> My system seems indeed more stable.

Therefore I saw that my link quality between coordinator and connected devices become lower than before.

So I did some tests with the same device. 
I tried several values for transmit_power setting and generated a network map for each of them.

With CC1352P2_CC2652P_other_coordinator_20210708 firmware : 

 - no transmit_power setting or transmit_power = 5, 10, 15 or 20 :  Same value for link quality ( around 25) with my test device.

[image]

 - transmit power = -5 : Link quality = 12

[image]
 

With Egony Firmwares  ( CC2652P_E72_20210901 and CC2652P_E72_20210211 )  : 

- transmit power =20

[image]

- transmit power = 5

[image]

- transmit power = -5

[image]

Conclusion : 
  --> Koenkk firmwares does not take into account transmit_power more than 5dbm

Can you please have a look on this ?

## What did you expect to happen

Koenkk firmwares take into account transmit_power more than 5dbm 

## How to reproduce it (minimal and precise)

## Debug info

Zigbee2MQTT version: Zigbee2MQTT version 1.21.1-dev commit: 0dfadb3a on RPI 3B+ Venv install
Adapter hardware: CC2652P E72-2G4M20S1E
Adapter firmware version: CC1352P2_CC2652P_other_coordinator_20210708.zip

Koenkk, 2021-09-26:
I'm aware of this, will fix it on next firmware release.

ALaDoffe, 2021-09-28:
Ok Well. Thx.
Do you know when the next firmware will be released ?

Koenkk, 2021-09-28:
TI release every quarter, last was 13 July 2021 so I expect it soon.

McGiverGim, 2021-09-29:
> TI release every quarter, last was 13 July 2021 so I expect it soon.

I don't know if this is what you are talking for, but your firmware uses SDK 5.10.00.48 according to the changelog and Egony has upgraded to SDK 5.20.00.52. This is the TI release?

Koenkk, 2021-09-29:
It is, but I experienced mem errors with `5.20.00.52` and therefore decided to skip it.

Shamshala, 2021-11-01:
Ok, so I wasn't crazy. It's a bit strange to see a weaker signal through CC2652P in compare to CC2530+CC2591 and without any reaction to power settings.

ALaDoffe, 2021-11-22:
Hello @Koenkk 
Does the new firmware from Z-Stack_3.x.0 20211114 correct this issue ?

Shamshala, 2021-11-22:
@ALaDoffe I gave it a try and seems like it's possible to increase the power but joining is broken somehow (I wasn't able to add anything new).

Koenkk, 2021-11-22:
@ALaDoffe not yet, I'm working on it

Koenkk, 2021-12-16:
These fws should allow 5dbm+ (with a max of 20dbm), default is 9dBm. Waiting for ITEAD (Sonoff) to confirm this since I don't have RF measurement equipment.
[CC1352P_CC2652P_TX_POWER.zip](https://github.com/Koenkk/zigbee2mqtt/files/7728605/CC1352P_CC2652P_TX_POWER.zip)

Hedda, 2021-12-16:
FYI, discussion about firmware with configurable RF output power for CC2652P and CC1352P here -> https://github.com/Koenkk/Z-Stack-firmware/issues/323

Koenkk, 2021-12-17:
Firmware 20211217 allow settings the transmit power > 5dBm (9dBm by default), ITEAD (Sonoff) confirmed this is working (using a RF analyzer). Link: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

closed 2021-12-17 by Koenkk

diginfo, 2022-10-21:
Hmm, I don't seem to be seeing any difference in LQI readings no matter which firmware I use, and no matter what the TX Power (transmit_power) settings in zigbee2mqtt in HA.

I have confirmed from the logs, that the command to change power setting is sent from zigbee2mqtt:

```
info  2022-10-21 08:47:00: Set transmit power to '2'
info  2022-10-21 08:49:03: Set transmit power to '20'
```

I have tried the the following versions and as far as I can tell there appears to be no change in the LQI for any of the connected devices or routers:

```
CC1352P2_CC2652P_launchpad_coordinator_20211217.hex  
CC1352P2_CC2652P_launchpad_coordinator_20220928.hex
CC1352P2_CC2652P_launchpad_coordinator_20220219.hex  CC1352P2_CC2652P_launchpad_coordinator_TX_POWER.hex
```

Am I missing something, and is anybody else successfully able to change transmit_power in zigbee2mqtt and see noticeable differences in LQIs ?

diginfo, 2022-10-21:
Ahh, now this makes a very noticeable increase in lqi: https://github.com/zigpy/zigpy-znp/issues/119#issuecomment-1001783291

But I cant get any changes using the configurable firmware above.

Koenkk, 2022-10-21:
LQI is from device -> coordinator; increasing TX power only improves the coordinator -> device LQI (which you cannot see)

diginfo, 2022-10-23:
Thanks so much for clarifying that.

RaNo99, 2022-11-15:
Hi @Koenkk I thought the reason why each link on the Zigbee network map shows two values is that the value before the slash is the very "coordinator -> device linkquality"... isn't it?