Koenkk/zigbee2mqtt #8893: Tuya Smoke Sensor support _TZE200_uebojraa

ramelito, 2021-09-27:
## Information about the device + link
https://aliexpress.ru/item/1005002842149725.html

## data/database.db entry of the device
..
{"id":37,"type":"EndDevice","ieeeAddr":"0xa4c1381da61e2aa1","nwkAddr":43326,"manufId":4417,"manufName":"_TZE200_uebojraa","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"]3�(\u00135\u0000\u0000\u0000e5\u0000\u0000\u0000\u0012","65506":34,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_uebojraa","powerSource":3,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1632734709594}

ramelito, 2021-11-28:
Any chance to get support for this sensor?

jrusi, 2021-12-06:
It looks the same as _TZE200_t5p1vj8r which is supported. Is there a way to tell Z2M to use the settings of TZE200_t5p1vj8r when connecting TZE200_uebojraa?

Koenkk, 2021-12-06:
@morphygee  you added this device in https://github.com/Koenkk/zigbee2mqtt/issues/9555 , could you help?

morphygee, 2021-12-07:
indeed... I do not see the battery level. 
but smoke tetection works without problems.

this is the part of my tuya.js:

`    {
        zigbeeModel: ['5p1vj8r'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t5p1vj8r'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_uebojraa'}],
        model: 'TS0601_smoke',
        vendor: 'TuYa',
        description: 'Smoke sensor',
        fromZigbee: [fz.tuya_smoke],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },
        exposes: [e.smoke(), e.battery_low()],
    },`

Koenkk, 2021-12-07:
@morphygee it doesn't seem to support a battery level (for that 1 should be in the `inClusterList` in database.db), it only supports `battery_low`.

jrusi, 2021-12-08:
I have exactly the same settings in tuya.js
When I connect new smoke sensor I can see this message:
`debug Device 'Czujka_dymu_sypialnia' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Czujka_dymu_sypialnia","ieee_address":"0xa4c138547b33c612"},"type":"device_announce"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Czujka_dymu_sypialnia"},"type":"device_announced"}'
debug Retrieving state of 'Czujka_dymu_sypialnia' after reconnect
debug Received Zigbee message from 'Czujka_dymu_sypialnia', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
debug Received Zigbee message from 'Czujka_dymu_sypialnia', type 'attributeReport', cluster 'genBasic', data '{"65506":49,"65508":1,"appVersion":67}' from endpoint 1 with groupID null
debug Received Zigbee message from 'Czujka_dymu_sypialnia', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,184,17,0,3,64],"type":"Buffer"}' from endpoint 1 with groupID null
debug Received Zigbee message from 'Czujka_dymu_sypialnia', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID null`

and in the exposed tab:
`smoke
Indicates whether the device detected smoke  N/A
battery_low
Indicates if the battery of this device is almost empty  N/A
linkquality
Connection quality (signal strength) 231lqi`

State tab:
`{
    "linkquality": 231
}`

ramelito, 2022-01-08:
> uebojraa

Hi, thanks.
However i do not see it supported in 1.22.1. Later release?

closed 2022-02-17 by github-actions[bot]