Koenkk/zigbee2mqtt #8900: network_key: GENERATE as default?

NikitaKorneev, 2021-09-28:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/6970

<div type='discussions-op-text'>

<sup>Originally posted by **NikitaKorneev** April  5, 2021</sup>
Being a newbie, I only now realized that my default network key allows any attacker to controll my home from anywhere near the house.

Its alright, ill change the key and re-pair all 40 of my devices...

I was wondering - why is the default config the way it is - why is the key:
   `- 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13`
and not `GENERATE`?

Are there any issues that prevent generation by default?</div>

I am not sure if there are any underlying issues with having `generate` as default, but it seems that will make default networks more secure....

elas123, 2021-10-02:
i dont believe u can use generate when using home assistant supervisor

NikitaKorneev, 2021-10-02:
@Koenkk is it really so - Home Assistant's config cannot have GENERATE in it?

Koenkk, 2021-10-03:
@ciotlosm https://github.com/Koenkk/zigbee2mqtt/issues/8900#issuecomment-932740243 could you check this?

ciotlosm, 2021-10-04:
Will take a look. Probably requires schema updates.

NikitaKorneev, 2021-11-07:
@ciotlosm I was wondering if there is any progress...

joshuaspence, 2021-12-09:
Not stale

NikitaKorneev, 2022-01-10:
not stale at all! I am still interested in knowing

closed 2022-02-17 by github-actions[bot]