Koenkk/zigbee2mqtt #8902: TuYa TS0601 thermostat shows redundant attributes for each sensor

groenmarsmannetje, 2021-09-28:
## What happened
The different sensors that are created for this device contain a lot of attributes that are redundant. In Home Assistant this duplicate information is just waste especially when the recorder also logs all this redundant information in the database.

## What did you expect to happen
Only include attributes for a device once and add it to the main component or to a sensor when it has relevance to that sensor.

## How to reproduce it (minimal and precise)
Look at the attributes of the several sensors that are created and see that they are a just copy of each other.

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: zzh
Adapter firmware version: latest

Plawasan, 2021-09-29:
You can get rid of those by disabling legacy attributes in Z2M config however be aware of this impact at the moment - https://github.com/Koenkk/zigbee2mqtt/issues/8807

groenmarsmannetje, 2021-09-29:
> You can get rid of those by disabling legacy attributes in Z2M config however be aware of this impact at the moment - #8807

Thanks, I did not know that. Will check if it works for me without any side effects.

groenmarsmannetje, 2021-09-30:
I tried it, but it did not work for me, because I use some attributes in my automations.
And as it removes all attributes from all entities, these automations do not work anymore.
So I had to enable legacy attributes again.

closed 2021-11-07 by github-actions[bot]

groenmarsmannetje, 2021-11-25:
Relates to #9269