Koenkk/zigbee2mqtt #8905: Option for IEEE State Topics

DanielPeinhopf, 2021-09-28:
It would be great if you could add a setting to specify how the device topics should look like.

At the moment the topics use the IEEE address by default but use the friendly name if a device has been renamed. I want to force using the IEEE address even if the device has been renamed.

The reason is that i want to use this address as the device key in my home automation controller so i can rename the devices in z2m whenever i want without needing to rename it in my controller too.

danieledwardgeorgehitchcock, 2021-09-29:
I think this is already the case - If you rename the device via the Zigbee2MQTT UI, there is an option called `Update Home Assistant entity ID` which if left unchecked, would only rename the device in Zigbee2MQTT and not Home Assistant

DanielPeinhopf, 2021-09-29:
> I think this is already the case - If you rename the device via the Zigbee2MQTT UI, there is an option called `Update Home Assistant entity ID` which if left unchecked, would only rename the device in Zigbee2MQTT and not Home Assistant

Thanks, but i'am not using Home Assistant, i'am using my own custom made Smart Home Server. So the HA integration is completely disabled in z2m.

closed 2021-11-07 by github-actions[bot]