Koenkk/zigbee2mqtt #8909: Trouble setting up device.

de-krabburger, 2021-09-29:
I have multiple Ikea lights working in my system. Now i'm trying to add the Ikea light LED1934G3. 

Since it was in de supported devices list i thought it would work right out of the box like the rest of the lights i use from Ikea. 

I can add the light and found the page 'how to support new devices' in the log. Looking into this i found the ikea.js file to add and started tho follow the instructions. I can't really figure out what i'm doing wrong or missing.
I tried looking into some other 'how to' examples by the interwebs but without succes. 

Normally i have a group of friends with more experiance who help me setup things i can't. This if totally new for them and they can't help me. 
I hope someone can help me with this. 
(This is the first time asking on a forum, if I miss something just say it).

danieledwardgeorgehitchcock, 2021-09-30:
Hi,

This device was added to the Dev codebase 19 days ago. The Dev branch is merged with the Prod branch monthly at the end / beginning of each month.

You can either switch to the Dev / Edge branch of Zigbee2MQTT or wait a couple more days and the device will become supported.

de-krabburger, 2021-09-30:
Oke Thank you. I guess I will wait a few days for this update. 

Is there a place were I can read the things you are telling me about?

danieledwardgeorgehitchcock, 2021-09-30:
To view the releases see here: [Releases](https://github.com/Koenkk/zigbee2mqtt/releases)

To view how to switch to the Dev branch see here: [Dev Branch](https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

For looking at when devices were added to the codebase, do a search in here: [Zigbee Herdsman Converters](https://github.com/Koenkk/zigbee-herdsman-converters)

de-krabburger, 2021-09-30:
I looked in the links you send. If i search for my type bulb in the releas notes it says that it has been added in july already. 

The commit you send is for the E26 version. I have the E27 bulb. I found another from june about [LED1934G3](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2741)

I hope it will work after the update. Otherwise i don't know what to do.

danieledwardgeorgehitchcock, 2021-09-30:
If you have access to the file system for your Zigbee2MQTT, you can open up the database.db file in a text editor (while the device is added) and search for the line containing your device. If you post it up here I’d be able to tell you if it’s going to work and if not, create a device handler for you in Dev :-)

johnvprcam24573455623, 2021-10-01:
It looks like it will be added to Prod soon:

https://github.com/Koenkk/zigbee2mqtt/issues/8909

> Hi,
> 
> A new version of this device was added to the Dev codebase 19 days ago. The Dev branch is merged with the Prod branch monthly at the end / beginning of each month.
> 
> You can either switch to the Dev / Edge branch of Zigbee2MQTT which will add support for your device, or wait a couple more days, update to the latest Zigbee2MQTT and try adding the device again.

de-krabburger, 2021-10-01:
I found out that there was another issue. Zigbee2Mqtt was not updated for a while. I didn't look at this because the auto update was active. 
A friend had the same issue and managed to fix it. The lights are working as they should now. 

Thank you for helping.

closed 2021-10-01 by de-krabburger