Koenkk/zigbee2mqtt #8911: Bseed (Led) Dimmer TS0601  _TZE200_3p5ydos3

jvrrr, 2021-09-29:
Hi, I finally found a Zigbee dimmer that fit in a Dutch walloutlet.
https://nl.aliexpress.com/item/1005003249382419.html

Only It is not supported by Zigbee2mqtt.:
sep 29 11:32:02 zigbee npm[1472]: Zigbee2MQTT:warn  2021-09-29 11:32:02: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_3p5ydos3'

"friendly_name":"0xa4c1386b241d352e","hardwareVersion":1,"ieeeAddr":"0xa4c1386b241d352e","lastSeen":1632908752773,"manufacturerID":4417,"manufacturerName":"_TZE200_3p5ydos3","model":"TS0601","modelID":"TS0601","networkAddress":46346,"powerSource":"Mains (single phase)","type":"Router","vendor":"-"}],"type":"devices"}'

Can you please add it?

## data/database.db entry of the device:
"id":46,"type":"Router","ieeeAddr":"0xa4c1386b241d352e","nwkAddr":46346,"manufId":4417,"manufName":"_TZE200_3p5ydos3","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\n(f\n(\u0012","65506":32,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_3p5ydos3","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1632908752773}

THANKS!!!!

Kind regards,

Jasper

jvrrr, 2021-10-08:
Hi, do you need some more information to fix this?
Thanks again!

jvrrr, 2021-11-01:
I have created a file TS0601.js and the device has been reconized:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '_TZE200_3p5ydos3', // Vendor model number, look on the device for a model number
    vendor: '_TZE200_3p5ydos3', // Vendor of the device (only used for documentation and startup logging)
    description: 'BSEED Dimmer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

jvrrr, 2021-11-01:
I add the following to my configuration.yaml and reboot zigbee2mqtt:

advanced:
  log_level: debug
external_converters:
  - TS0601.js

I hit te buttons a couple off time:

Down Button:

nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,3,197],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":59}' from endpoint 1 with groupID 0
nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[0,0,3,197],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":59}'
nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":60}' from endpoint 1 with groupID 0
nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":60}'
nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":61}' from endpoint 1 with groupID 0
nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":61}'
nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":62}' from endpoint 1 with groupID 0
nov 01 21:24:29 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:29: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":62}'
nov 01 21:24:30 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:30: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":63}' from endpoint 1 with groupID 0
nov 01 21:24:30 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:30: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":63}'
01 21:24:31 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:31: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":64}' from endpoint 1 with groupID 0
nov 01 21:24:31 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:31: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":64}'
nov 01 21:24:31 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:31: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":65}' from endpoint 1 with groupID 0
nov 01 21:24:31 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:31: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":65}'
nov 01 21:24:32 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:32: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":66}' from endpoint 1 with groupID 0
nov 01 21:24:32 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:24:32: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":66}'

UP Button:
nov 01 21:27:50 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:27:50: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":67}' from endpoint 1 with groupID 0
nov 01 21:27:50 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:27:50: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":67}'
[... cut ...]
nov 01 21:32:52 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:32:52: Received Zigbee message from '0xa4c1386b241d352e', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":82}' from endpoint 1 with groupID 0
nov 01 21:32:52 zigbee npm[1557]: Zigbee2MQTT:debug 2021-11-01 21:32:52: No converter available for '_TZE200_3p5ydos3' with cluster 'manuSpecificTuya' and type 'commandGetData' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"status":0,"transid":82}'

jvrrr, 2021-11-01:
Can someone please help me out with the next step....I do not understand how to fix this.

Thanks!

markjansen1973, 2022-01-15:
Hi all,
I suffer with the same problem, is there already a fix known?

thanks in advance,
Mark

oberschwabe, 2022-01-26:
Hi all,
same here - how could this be moved ahead?

Bernd

jvrrr, 2022-01-27:
Nope, still waiting on a reaction.....
Maybe you can make a new topic?!

Koenkk, 2022-01-27:
Does it work with the following external converter?

```js
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_3p5ydos3'},
    ],
    model: 'TS0601_dimmer',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.tuya_dimmer_state, tz.tuya_dimmer_level],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
    whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
    ],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m and try to control the device from the z2m frontend

jvrrr, 2022-01-29:
@Koenkk 
First I want to thank you, what a great work!

I tried  out your code and it almost work perfectly, but adjust brightness by z2m is not working :

On/OFF:
- z2m on/off, the dimmer goes on/off = okay
- manually on/off, in z2m the status change = okay

Dimming:
- z2m dimming up/down, nothing change=not okay
- Manually dimming up/down, status update in z2m works=okay

What is the next step to find out what goes wrong?

Thanks a lot!

Koenkk, 2022-01-29:
@jvrrr that complicates things, this will require some reverse engineering when using the device with the original gateway, https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats might help

jvrrr, 2022-01-29:
@Koenkk 
I just found out something and maybe this can help me out?!.

If I set in z2m the dimmer on 50%, then is that the lowest level I manually can dim by hand .. so I can't dim any lower then 50%.
So I must set it to 0% in z2m , so I'm able to dim it manually to 0%.

Is this something you have ever seen?

Koenkk, 2022-01-30:
@jvrrr so if I understand correctly, publishing to `zigbee2mqtt/MY_DEVICE_FRIENDLY_NAME`:

- `{"brightness":255}`: sets the max brightness
- `{"brightness":128}`: sets the min brightness
- `{"brightness":50}`: doesn't produce a lower brightness than 128

jvrrr, 2022-01-30:
Sorry for not being clear, what I ment is:
[image]
When I set this  (z2m-> device->exposes) on 50%, I 'm not able to manually set the brightness lower then 50%. (So I'm not able to dim the lamp down to 10% by hand anymore)
I sounds more now the option is turnt in to in Brightness minimum value.

jvrrr, 2022-01-30:
Log from z2m:
Debug Received MQTT message on 'zigbee2mqtt/0xa4c138f78de31872/set' with data '{"brightness":226}'

Debug Publishing 'set' 'brightness' to '0xa4c138f78de31872'

Debug Received Zigbee message from '0xa4c138f78de31872', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,3,118],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"seq":1}' from endpoint 1 with groupID 0

Info MQTT publish: topic 'zigbee2mqtt/0xa4c138f78de31872', payload '{"brightness":225,"level":886,"linkquality":102,"state":"ON"}'

Debug Received MQTT message on 'zigbee2mqtt/0xa4c138f78de31872/set' with data '{"brightness":82}'

Debug Publishing 'set' 'brightness' to '0xa4c138f78de31872'

Debug Received Zigbee message from '0xa4c138f78de31872', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,66],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"seq":1}' from endpoint 1 with groupID 0

Info MQTT publish: topic 'zigbee2mqtt/0xa4c138f78de31872', payload '{"brightness":80,"level":322,"linkquality":94,"state":"ON"}'

Koenkk, 2022-01-31:
Can you try with this external converter?

```js
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');

const tzLocal = {
    tuya_dimmer_level: {
        key: ['brightness_min', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            // upscale to 1000
            let newValue;
            let dp = tuya.dataPoints.dimmerLevel;
            if (meta.device.manufacturerName === '_TZE200_3p5ydos3') {
                dp = tuya.dataPoints.eardaDimmerLevel;
            }
            if (key === 'brightness_min') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                    dp = tuya.dataPoints.dimmerLevel;
                } else {
                    throw new Error('Dimmer brightness_min is out of range 0..100');
                }
            } else if (key === 'level') {
                if (value >= 0 && value <= 1000) {
                    newValue = Math.round(Number(value));
                } else {
                    throw new Error('Dimmer level is out of range 0..1000');
                }
            } else if (key === 'brightness_percent') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness_percent is out of range 0..100');
                }
            } else {
                if (value >= 0 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 0, 255, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness is out of range 0..255');
                }
            }
            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
        },
    },
};

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_3p5ydos3'},
    ],
    model: 'TS0601_dimmer',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.tuya_dimmer_state, tzLocal.tuya_dimmer_level],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
    whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
    ],
};

module.exports = definition;

```

jvrrr, 2022-02-01:
@Koenkk 
I have tried it out, same issue.

debug Received MQTT message on 'zigbee2mqtt/0xa4c138f78de31872/set' with data '{"brightness":196}'
debug Publishing 'set' 'brightness' to '0xa4c138f78de31872'
debug Received Zigbee message from '0xa4c138f78de31872', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,3,1],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"seq":1}' from endpoint 1 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/0xa4c138f78de31872', payload '{"brightness":195,"level":769,"linkquality":149,"state":"ON"}'

This device has got a other ID because I have two off them, and I want to test if it was a issues with the device it self.
So both the device has got the same issue.

I hope you have got a other trick  ;-)

Thanks a lot!!

Koenkk, 2022-02-02:
@jvrrr then this device uses a different data point for the brightness, you can find out which via https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

jvrrr, 2022-02-05:
@Koenkk 
I just bought a Tuya gateway...I will have to wait till it is delivered.
Thanks so far!

Aledan862, 2022-02-08:
@Koenkk I have the same dimmer. So, when adding a converter from your last post dimmer start working completely fine.  Now it's turning on/off and can change brightness.

amaciuc, 2022-02-14:
> @Koenkk I just bought a Tuya gateway...I will have to wait till it is delivered. Thanks so far!

I am waiting news from your discover

jvrrr, 2022-02-15:
> > @Koenkk I just bought a Tuya gateway...I will have to wait till it is delivered. Thanks so far!
> 
> I am waiting news from your discover

It takes a couple of weeks till it will arrive.
You have the same issue?

jvrrr, 2022-02-25:
@Koenkk 
I finally received the Tuya bridge....
The manual on the Zigbee2mqtt is not matching with how it looks now on the Tuya site and also the Google dev tools has been changed.....so I hope I have found the right information:

{startRowId: "", pageNo: 1, pageSize: 10, code: "2", startTime: 1645644701499, endTime: 1645817501500,…}
code: "2"
deviceId: "bfacf17cffaa8e81ef74qf"
endTime: 1645817501500
pageNo: 1
pageSize: 10
pageStartRow: ""
projectCode: "p1645127508856sqrgta"
region: "EU"
sourceId: "eu16451295720384SW1W"
sourceType: "4"
startRowId: ""
startTime: 1645644701499 

Thanks a lot, would be great if this will work out!

jvrrr, 2022-02-25:
@Koenkk

Sorry Koen,  I have been stupid and forget a enter...............your script did work!!!:

--------------------------------------------------------------------------------------------------------------------------
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');

const tzLocal = {
    tuya_dimmer_level: {
        key: ['brightness_min', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            // upscale to 1000
            let newValue;
            let dp = tuya.dataPoints.dimmerLevel;
            if (meta.device.manufacturerName === '_TZE200_3p5ydos3') {
                dp = tuya.dataPoints.eardaDimmerLevel;
            }
            if (key === 'brightness_min') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                    dp = tuya.dataPoints.dimmerLevel;
                } else {
                    throw new Error('Dimmer brightness_min is out of range 0..100');
                }
            } else if (key === 'level') {
                if (value >= 0 && value <= 1000) {
                    newValue = Math.round(Number(value));
                } else {
                    throw new Error('Dimmer level is out of range 0..1000');
                }
            } else if (key === 'brightness_percent') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness_percent is out of range 0..100');
                }
            } else {
                if (value >= 0 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 0, 255, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness is out of range 0..255');
                }
            }
            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
        },
    },
};

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_3p5ydos3'},
    ],
    model: 'TS0601_dimmer',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.tuya_dimmer_state, tzLocal.tuya_dimmer_level],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
    whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
    ],
};

module.exports = definition;
-------------------------------------------------------------------------------

Can you please add it as new supported device.

You are great ,thanks a lot!!!!!!!

Koenkk, 2022-02-26:
Done!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-26 by Koenkk

moshed, 2022-12-16:
@jvrrr are you able to toggle off this switch? I. have same on and am having some weird issues

dmak, 2023-02-07:
@jvrrr Maybe you could share with us the following information:
* Does this dimmer correctly functions as Zigbee router, i.e. extends the Zigbee network?
* Does it have a min/max brightness setting? E.g. some LED lamps start to blink on max power, but if it is limited then LED does not blink.