Koenkk/zigbee2mqtt #8914: Forcibly deleted devices stay in configurator

okastl, 2021-09-29:
## What happened
When a device is forcibly removed, it is deleted from database, but stays in the configurator.  Over time, coordinator_backup.json is filled with "dead" entries.

## What did you expect to happen
database.db and coordinator_backup.json should stay in sync, if a device is "forcibly removed".

## How to reproduce it (minimal and precise)
Add a device.
Force remove it.
Shut down z2m.
The delted device entry is present in  coordinator_backup.json

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC26X2R1
Adapter firmware version: 20210708

MokeleM, 2021-10-05:
Same issue here

okastl, 2021-11-05:
I recently noticed, that sometimes devices, which are "normal" deleted, stay in coordinator. Zigbee 3 devices (with the "link_key") are in the backup, but the "link_key" is removed.
Further, sometimes non-Zigbee 3 devices added are not added to the coordinator backup. Not tragic, as these devices don't need the additional encryption keys, but still, something is broken.

okastl, 2021-11-05:
> ## Debug info
> Zigbee2MQTT version: 1.21.1

The problem is the same in 1.22.0-dev commit: a09aa42

Koenkk, 2021-11-05:
This is expected, force remove just removes the device from the z2m db, not from the network.

okastl, 2021-11-05:
> This is expected, force remove just removes the device from the z2m db, not from the network.

Unfortunately this happens with "normal" remove, too. Just not as often and therefore not easy to reproduce.

okastl, 2021-11-08:
> Further, sometimes non-Zigbee 3 devices added are not added to the coordinator backup. 

To demonstrate the discrepancy I have made a little quick hack in python. It alerts you about differences between coordinator_backup.json and database.db.
After adding an CLA60 TW OSRAM Bulb, the device was not added to coordinator_backup.json, and my tool showed this error:

ieeKey 7cb03eaa00... (Osram) missing!

I have attached the Python program, maybe it is helpful. It expects devices in a separate file.
Without such a tool missing or non deleted entries in coordinator_backup.json are not easily spotted.

[z2mcheck.zip](https://github.com/Koenkk/zigbee2mqtt/files/7499588/z2mcheck.zip)

ptath, 2021-11-29:
Any ideas how to fix this?

closed 2022-01-06 by github-actions[bot]